linuxwacom/xf86-input-wacom #209: CTL-480 Eraser not working

Perangelot, 2021-12-16:
I have a Wacom CTL-480 tablet which works quite well out of the box. I use KDE Neon and have added the KDE module for wacom tablets. Now there is two problems:

1. xsetwacom recognizes an eraser although my stylus does not have one.
2.  I cannot assign the eraser function to button three, so that, irrespective of the program I use, pressing and holding that button calls the eraser. The KDE module only allows to set the mous-clicks (for some reason, to 32 of them).

According to the offical ubuntu website, CTL-480 should work out of the box. In Windows, I can just use the wacom program to define it globally or program-specific. I have tried virtually anything I found on the net and am starting to despair. I have just started using Linux. Can anyone help me?

jigpu, 2021-12-16:
> xsetwacom recognizes an eraser although my stylus does not have one.

This is normal behavior. We cannot detect if your stylus has an eraser or not so we automatically create an eraser device "just in case."

> I cannot assign the eraser function to button three, so that, irrespective of the program I use, pressing and holding that button calls the eraser. The KDE module only allows to set the mous-clicks (for some reason, to 32 of them).

If I understand you correctly, this is also normal behavior. There is no way to assign one of the side-switches to an "erase" function. There has been some thought about how to support this feature (and its opposite -- re-assigning dedicated eraser buttons to just send normal mouse clicks instead), but its unfortunately pretty challenging. The switches can, however, be assigned to send various mouse clicks or keyboard events, which individual applications may use to switch to their eraser mode.

whot, 2021-12-21:
#134 is the issue regarding remapping the eraser button but long story short: I couldn't make this work in libinput, the heuristics needed are very complex. And it's even less likely to be possible in the wacom driver which is harder to change for things like that.

Pinglinux, 2022-08-04:
To X Input, stylus and eraser are two independent tools. We can either use the tool as a stylus or an eraser. But, we can't use the tool as a stylus and an eraser at the same time. That's the challenge of assigning a button as an eraser since it is like making the stylus as an eraser.

To have the eraser feature, we need a pen that supports eraser. This is an unique case for Linux. We got two tools from one physical pen, as long as the pen has eraser. That's how it works on Linux.

closed 2022-08-04 by Pinglinux