linuxwacom/xf86-input-wacom #216: Elan 2a70 doesn't have an eraser

MSmolarz, 2022-01-13:
Hi,
on my HP ProBook X360 11 G5 EE the Stylus got recognized and works, but there is also a Eraser recognized which doesn't exists. So the Stylus does not have an extra eraser. This lead to issues in Xournal++ and other programs using this feature.
This is the output of `xinput list`:
```
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ xwayland-pointer:16                     	id=6	[slave  pointer  (2)]
⎜   ↳ xwayland-relative-pointer:16            	id=7	[slave  pointer  (2)]
⎜   ↳ xwayland-touch:16                       	id=9	[slave  pointer  (2)]
⎜   ↳ xwayland-stylus:16                      	id=10	[slave  pointer  (2)]
⎜   ↳ xwayland-eraser:16                      	id=11	[slave  pointer  (2)]
⎜   ↳ xwayland-cursor:16                      	id=12	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ xwayland-keyboard:16                    	id=8	[slave  keyboard (3)]
```
I've uploaded the tarball by sysinfo scipt here: https://github.com/linuxwacom/wacom-hid-descriptors/issues/185
and my .tablet description so that device get recognized in the gnome settings here: https://github.com/linuxwacom/libwacom/issues/465

Is it possible to fix this issue in the xf86-input-wacom driver or do I need to approach the problem differently??
Thanks
Mark

whot, 2022-01-17:
first a note - when you're using Wayland you're not using this driver at all so this issue is definitely not a wacom driver issue.

The issue in this particular case is in Xwayland which unconditionally adds those devices (similar to the wacom driver though) even though they may never be used. This *could* be fixed but it's a bit of effort in Xwayland and unlikely to be implemented unless you yourself send the patches.

Having said that, the wacom driver has initialized these devices for over a decade now, I'm surprised that Xournal has buggy behaviours there. What exactly is buggy here?

closed 2022-01-17 by whot

MSmolarz, 2022-01-20:
Yes i see, i'm using wayland.
But I've seen, that there is a file named libwacom.stylus. If I understand it correctly there are a lot of pens and their specifications described in it. Is that correct? Wouldn't my problem be solved if my pen was added there?

whot, 2022-01-20:
it wouldn't help much due to technical limitations. the vast majority of the styli in that file can be identified due to unique IDs that wacom builds into its professional stylus range. That is not the case for the elan devices where we just fall back to a set of generic styli. In https://github.com/linuxwacom/wacom-hid-descriptors/issues/185 the hid report descriptor includes the `Invert` usage which is the eraser - this means that technically your tablet is capable of handling an eraser even if your current pen doesn't have one.

Since those panels tend to be re-used across multiple devices we can't just limit ourselves to the pen device, it'll just break on the next device they're integrated in.