linuxwacom/xf86-input-wacom #220: Synaptics-like smoothness for PanScrolling

Greenscreener, 2022-01-30:
I have a One by Wacom M, that I use the PanScrolling feature on. (`xsetwacom set "..." Button 2 pan`).
This feature is great and allows me to navigate the UI of almost all programs very naturally. 
There's however one thing, that annoys me slightly: For panscrolling to be usable, you have to set the PanScrollThreshold really high (it's 150 for me) which means the pen moves 150 dots, sends a single large scroll event, another 150 dots, another large scroll event, which makes scrolling really choppy. When I scroll on my touchpad, scrolling is much smoother and doesn't happen in (noticeable) discrete steps. 

The behaviour demonstrated in xournalpp:

https://user-images.githubusercontent.com/22398555/151708200-518ad2f4-b81d-42e6-96a4-7e2c576c3931.mp4

If velocity could be implemented as well, that would be a very nice touch, but that isn't that useful for tablets as it is useful for touchpads. 

After doing some research, this is apparently done using valuators 2 and 3 in an X motion event. I will attempt to implement this myself and post a PR, but I'm not very experienced, so we'll see how that goes.

Greenscreener, 2022-01-30:
Alright, update, I thought it was going to be a much easier modification. I found that valuators aren't actually global (as in 2=scrollX) but they have to be initialized individually and somehow given properties and advertised to X. If anybody could tell me how to do that properly, I'll be glad to implement it myself, but as I said, I have no experience with writing drivers, so I probably won't be able to do it. If I could help with this in any way, let me know.

whot, 2022-01-31:
The [xf86-input-libinput](https://gitlab.freedesktop.org/xorg/driver/xf86-input-libinput/) driver is probably the simplest driver to understand because all the device handling is done in libinput and the driver is just a middle layer. With the recent wacom driver rework, the bits in `src/x11/` work very similar to the xf86-input-libinput driver in terms of high-level architecture. The files in `src` are the core driver and independent of X, the bits in `src/x11` convert to X.

The main problem with the wacom driver atm is that **it does not have a scrolling axis**. All scroll events are sent as legacy button events 4/5 (vertical) and 6/7 (horizontal).

To get anything smooth like touchpads you need one valuator designated as scroll valuator and then send **motion events** with that valuator set. Grep for `SetScrollValuator` in the xf86-input-libinput code to see how that works and `xf86libinput_handle_axis` for the handling of the scroll events. These are the bits that need to be replaced in this driver to get smooth scrolling.

Greenscreener, 2022-11-23:
This has been resolved in #222 and #298.

closed 2022-11-23 by Greenscreener