linuxwacom/xf86-input-wacom #221: 1.0 release?

whot, 2022-01-31:
I'm planning to get the 1.0 release out in the next few days. 1.0 because we're at 0.40 right now, the driver has been "stable" for years so we might as well drop the 0 prefix. Unless someone else has a creative suggestion for version numbers.

@Pinglinux, @jigpu - anything missing that I need to wait for?

Pinglinux, 2022-01-31:
@whot I believe Jason has some patches to merge before the release. Hopefully he'll send the pull request soon.

Thank you for taking care of xf86-input-wacom. Tagging the release as 1.0 works for me.

jigpu, 2022-01-31:
My patchset grew a bit more than expected, so since they're just cleanups I'm going to put them on hold. I don't want to invalidate all the testing Ping did previously. Go ahead with the big 1.0 if you'd like :laughing:

whot, 2022-01-31:
thanks! plus, this way we can blame any bugs on ping's testing ;)

whot, 2022-02-09:
So, after a bunch more build system fixes, we should be ready to go. At this point #243 is the only one that's a must-have, I don't have anything else urged queued up.

whot, 2022-02-17:
1.0 is out! goeth forth and thelebrathest!

closed 2022-02-17 by whot