linuxwacom/xf86-input-wacom #226: Absolute values passed to valuators in relative mode

Greenscreener, 2022-02-02:
While working on #222, I found a bug which I originally assumed to be related to my changes, but it's actually present in master. 

When in relative mode, it seems that absolute values are passed to the valuator even in relative mode, which makes the cursor fly off to the bottom edge of the screen. I bisected the problem and it seems to be introduced in 58a931bc21d1e78629a0873beb221252d065fdca.

Greenscreener, 2022-02-02:
A workaround that seems to work is removing these two [lines]( https://github.com/linuxwacom/xf86-input-wacom/blob/master/src/wcmCommon.c#L700-L701). I haven't been able to find out why or how as of yet.

whot, 2022-02-02:
Referencing #221, need to fix this before the release. Good find, thank.

whot, 2022-02-03:
This doesn't appear to happen in the core driver:
```
$ sudo ./builddir/wacom-record  --options "Mode=Relative" /dev/input/event4
    - source: 0
      event: motion
      axes: { x:    13, y:     7, pressure:    0, tilt: [  0,  0], rotation:   0, throttle:   0, wheel: -108, rings: [  0,   0] }
    - source: 0
      event: motion
      axes: { x:    15, y:     9, pressure:    0, tilt: [  0,  0], rotation:   0, throttle:   0, wheel: -108, rings: [  0,   0] }
    - source: 0
      event: motion
      axes: { x:    19, y:     9, pressure:    0, tilt: [  0,  0], rotation:   0, throttle:   0, wheel: -108, rings: [  0,   0] }
    - source: 0
      event: motion
      axes: { x:    20, y:     9, pressure:    0, tilt: [  0,  1], rotation:   0, throttle:   0, wheel: -108, rings: [  0,   0] }
    - source: 0
      event: proximity
      proximity-in: false
      axes: { x:     0, y:     0, pressure:    0, tilt: [ 30, 30], rotation:   0, throttle:   0, wheel: -900, rings: [  0,   0] }
```

I ran the xorg driver with master from today (be159bcd06c39ee1839efa66b732c810b6afe186) and I'm unable to reproduce with that too. It's a bit suspicious that we're posting all zeros (the mask is set for those when it probably shouldn't be) but i think that's what the driver has always done (confirmed with wacom 0.40.0 at least).

jigpu, 2022-02-04:
I've just pushed a 'fix-226' branch up to my repository which should fix the issue. Would you mind giving it a try? It looks like the problem was a result of us not moving the call to `wcmRotateAndScaleCoordinates` to before the point we save x and y in our axis struct.

https://github.com/jigpu/xf86-input-wacom/commits/fix-226

Greenscreener, 2022-02-05:
Sorry for the delay, but yes, it works correctly. Btw my setup is this:
```
Button 2 pan
PanScrollThreshold 150
Mode Relative
Area 0 0 36800 27000
```

PR #243 (2022-02-10): Rotate / scale coordinates before storing them in axes
The WacomAxisData calculated by `wcmSendEvents` is expected to contain
the X,Y coordinates which have been adjusted for device rotation and
scale. Prior to the refactoring, the values of X and Y (along with
everything else) were only packaged up for relaying at the very end
of the function. Now this packaging occurs much earlier which requires
our call to `wcmRotateAndScaleCoordinates` to move as well.

Fixes: 58a931bc21d1 (Abstract the event interface to pass a struct with axis data around )
Link: https://github.com/linuxwacom/xf86-input- [... cut]