linuxwacom/xf86-input-wacom #233: Add CI job with minimum meson version

whot, 2022-02-07:
Add a CI job that builds the driver with the minimum meson version we require in `meson.build` to make sure we don't rely on new features without knowing.

eli-schwartz, 2022-02-07:
In theory, Meson tries very hard to tag all new features with the version they were added in, and report warnings when you rely on too new features.

These warnings will become errors if you build using `--fatal-meson-warnings`, which is probably safe to enable because the current CI builds do not emit any meson warnings.

(If you do find new features that don't print a warning, please report them as a Meson bug! We will add the missed warning.)

whot, 2022-02-07:
ftr, in this particular case the warning was a "future deprecated" warning that prompted this issue, so meson did the right thing. I'll experiment with `--fatal-meson-warnings`, thanks!

eli-schwartz, 2022-02-08:
Ah yeah, those won't ever trigger fatal warnings, they are purely informational to tell you that you have the opportunity to upgrade some stuff if and only if you upgrade your minimum requirements.

Once you do upgrade your requirements the "future deprecated" gets upgraded to a "right now deprecated" and will trigger `--fatal-meson-warnings`. :D

...

Hmm, while you are adding additional jobs to test against a range of Meson versions, what are your thoughts on having a job that tests against `pip install --pre meson`, i.e. when a new release candidate of Meson is uploaded as a prerelease.

It's always good to have more testing of those, and if it does fail, then you can report a Meson regression bug before it has a chance to be released into the wild.

whot, 2022-02-09:
> what are your thoughts on having a job that tests against pip install --pre meson, i.e. when a new release candidate of Meson is uploaded as a prerelease.

I added this to some repo already (forgot which one. pipewire?) but for this driver it'll be overkill. The driver is quiet at times to there's a reasonable chance that we miss release candidates completely because nothing is built in that time frame.

eli-schwartz, 2022-02-09:
Alright, makes sense. :)

whot, 2022-02-11:
sadly enough, `--fatal-meson-warnings` has become a lot less useful given than our CI uses custom install directories. Any prefix other than `/usr` (or whatever it will be on ubuntu, only tested on Fedora) will now trigger this warning:
```
WARNING: Python files installed by Meson might not be found by python interpreter.
 This warning can be avoided by setting "python.platlibdir" option.
```
Sure, we could supply more options and I understand why the warning was added, but still...

closed 2022-02-11 by whot

eli-schwartz, 2022-02-11:
That warning is honestly pretty annoying, but meson 0.62 should hopefully have a single option that both suppresses the warning and makes the files be installed into a reasonable *autodetected* place. Maybe we can ultimately kill off the warning entirely.

It's a pretty silly hack for Debian's broken python, anyway.

PR #235 (2022-02-11): CI: add a job to build with the exact meson version we require
Just to catch any inadvertent reliance on features that came after our
minimum version. Meson *should* warn about this but let's be sure.

Fixes #233

cc @Pinglinux
commit c8e597aa (2022-02-11): CI: add a job to build with the exact meson version we require

Just to catch any inadvertent reliance on features that came after our
minimum version. Meson *should* warn about this but let's be sure.

Fixes #233