linuxwacom/xf86-input-wacom #247: Panscrolling doesn't work in QT apps

Greenscreener, 2022-02-13:
This is another issue I found while working on #222. Some debug logs can be found [here](https://github.com/linuxwacom/xf86-input-wacom/pull/222#issuecomment-1030687835)
I'll look further into it and bisect it tomorrow.

andrejp88, 2023-08-08:
I've been using my tablet more recently, and have noticed this problem with Qt apps. I think I've tested every Qt app I have installed.

The Dolphin Emulator (`dolphin-emu` in Arch), which uses Qt6, does not have this problem. However, qBittorrent and Stellarium, my only other Qt6 apps, *do* have the problem.

All other Qt-based apps on my computer depend on Qt5, not 6, and all of them have the scrolling issue, including Dolphin (the file explorer), system settings, and Plasma Desktop stuff such as system tray popups.

For some reason, the Dolphin Emulator does something different here.

Not sure if this helps at all but I thought it would be a bit more useful than just a thumbs up :stuck_out_tongue:

Pinglinux, 2024-05-18:
The tomorrow happened 2+ years ago. @Greenscreener must have been tied to other high priority tasks. I am closing the issue unless someone else has a solution for it.

closed 2024-05-18 by Pinglinux

UlyssesZh, 2024-05-18:
Then just leave it open.