linuxwacom/xf86-input-wacom #257: Allow negative panscrolling speed pwease UwU

MikuChan03, 2022-03-15:
Heya!

I'm in a bit of a bad mood so I'll spare you the whole Miku-Chan bit. Anyways, I'm one of the borderline insane people that like "unnatural" scroll, as libinput puts it, and want the screen contents to move upwards when my pen moves upwards. This of course works nicely with setting a negative panscrolling speed. Not wanting to do that at every reboot, I put it in a x11 config file. This also works tremendously, when setting a positive value that is. The Xserver is alright with me setting the speed to for example "1234". If I were to set it to "-1234" though, it goes like, awww, poor guy, lemme just fix this silly input mistake you make, and reverts it back to 1300 or 2600. I am very appreciative of you guys looking out to me to this extent, and I do I feel like I should at least send you guys a christmas card or whatever. Due to countless cerebral insufficientcies of mine, at times I really truly honestly would prefer a negative panscrolling speed, right from the getgo. So if one of you nice senpais would be so kind as to entertain this insufferable foolishness of mine, I would be boundlessly, everlastingly grateful.

Thank!

It goes like this (viewer discretion is advised):
```
[  2683.739] (--) wacom: GAOMON Gaomon Tablet Pen stylus: using pressure threshold of 851 for button 1
[  2683.739] (**) wacom: GAOMON Gaomon Tablet Pen stylus: panscroll is -1234
[  2683.739] (**) wacom: GAOMON Gaomon Tablet Pen stylus: panscroll modified to 2600
[  2683.739] (--) wacom: GAOMON Gaomon Tablet Pen stylus: maxX=33020 maxY=20320 maxZ=8191 resX=200000 resY=200000  tilt=enabled
```

Also, if Senpai Hutterer where to read this. Thanks for  believing in me, man. I asked you a while back how to set relative cursor movement with libinput and you said that while it sadly doesn't work, I can implement it if I want to. While I didn't do that, I actually added native support for some newer xppen tablet to digimend and sent the patch of the mr. Kosina and mr. Tissoire. They haven't had a look at it yet, but it's nothing to crazy and I have hopes for it getting accepted. So thanks for your encouragement! Also thanks for implementing panscroll at all.

MikuChan03, 2022-03-18:
I really don't want to bother anyone, but all I'm suggesting is that we might go:
```c
wcmLog(priv, W_CONFIG, "panscroll is %d\n", common->wcmPanscrollThreshold);
if (common->wcmPanscrollThreshold == 0) {
        common->wcmPanscrollThreshold = common->wcmResolY * 13 / 1000; /* 13mm */
}
if (common->wcmPanscrollThreshold == 0) {
        common->wcmPanscrollThreshold = 1000;
}
wcmLog(priv, W_CONFIG, "panscroll modified to %d\n", common->wcmPanscrollThreshold);
```
in wcmCommon.c
The == 0 part vs originally < 1, therefore eliminating the negative values some desire.
Is this at all reasonable?
@whot

MikuChan03, 2022-03-29:
Pwease?

jigpu, 2022-04-04:
Apologies for the delay... I think the idea of allowing a negative panscroll speed seems reasonable, particularly for opaque tablets where it would be analogous to inverted touchpad scrolling.

Have you tested the suggested change, @MikuChan03? If you have a working patch, I'd be happy to review a pull request for this feature. Otherwise I can add this to my personal to-do list and work on implementing it in between some other priorities...

jigpu, 2022-04-07:
I've just pushed a branch with the requested change to my repository. Would you mind testing it out, @MikuChan03? You can get a copy by running `git clone https://github.com/jigpu/xf86-input-wacom.git -b issue-257` and [follow our build instructions](https://github.com/linuxwacom/xf86-input-wacom/wiki/Building-The-Driver) to compile and install that branch from source. It should let you assign a negative panscroll threshold, resulting in the inverted scrolling behavior you desire.

MikuChan03, 2022-04-12:
@jigpu 
Hello!
So I tested it for like 0, -0, 0.1, 0.9, -0.1, -0.9 and the desired value of like 1300, -1300, 2600, -2600. Everything works as it does with upstream version, with the only difference being that negative values are not overridden. Everything is hunky-dory, real peachy creamy.

Maybe merge please?
Thank!

Pinglinux, 2022-04-14:
The patch looks good to me. It has been merged.

closed 2022-04-14 by Pinglinux

MacroHard-Winux, 2023-06-28:
nwice

pablodz, 2023-07-12:
O o

yuuuuuke, 2023-08-04:
> Owo

you are NOT mikuchan03 😭