linuxwacom/xf86-input-wacom #259: Question: Can I setting up wheel/ring mode at Intuos 4?

argxentakato, 2022-03-18:
On the Intous 4, you can switch between modes 1-4 by pressing the button in the centre of the ring.
However, xsetwacom does not seem to be able to change its behaviour depending on the ring mode. Pressing the button switches the LED, but there is no option to change the behaviour.

Am I just missing a setting or is it not possible to set up this ring mode with linux wacom alone?

My environment is as follows:

OS: Arch Linux
Kernel: 5.15.28-LTS
libwacom: 2.1.0
xf86-input-wacom: 1.0.0

Thanks.

Pinglinux, 2022-03-19:
The ring mode change is more than a xsetwacom command. It should be controlled by the system's desktop control panel. Please take a look at this link https://ubuntuforums.org/showthread.php?t=1380744&page=32 to see how for you can get.

Pinglinux, 2022-03-19:
Close this request since it is not a driver issue

closed 2022-03-19 by Pinglinux