linuxwacom/xf86-input-wacom #267: Add an option to a mouse mode

mhalano, 2022-05-15:
Windows software has a mouse mode available. I think we could implement a mouse mode as well, and maybe it can be available easily using the interface (GNOME and KDE). What do you think?

whot, 2022-05-16:
Can you please specify what you mean exactly by "mouse mode"?

mhalano, 2022-05-16:
There is an excerpt of documentation [here](https://101.wacom.com/UserHelp/en/MappingMouseMode.htm)

Pinglinux, 2022-05-16:
@mhalano That's already supported in Gnome Control Center through "Wacom Tablet" settings and xsetwacom through
                      Mode             - Switches cursor movement mode (default is absolute)
You need to set the tool in relative mode if xsetwacom works for you or set "Tracking Mode" to Touchpad (relative) in Gnome Wacom Tablet.

closed 2022-05-31 by Pinglinux