linuxwacom/xf86-input-wacom #271: Assign Buttons Wacom Intuos Pro M

crchmatefis, 2022-05-28:
After installing Ubuntu 22.04(wayland and xorg), I tried to assign keystrokes to the buttons of my wacom tablet, but, the system does not let me select the option, in general it does not let me select any of the options. Has anyone an idea how to fix it?

Note: in ubuntu 20.04 I could assign keystrokes to each button on my tablet.

whot, 2022-05-30:
Are you running Wayland or X11? If the former, it's not this driver. Run `xisxwayland -v`, if it says YES then you're on wayland.

crchmatefis, 2022-06-04:
In ubuntu 22.04 either running Wayland or X11 I can't assign a keystroke to any button on my tablet... in ubuntu 20.04 with X11 I could.

crchmatefis, 2022-06-04:
Hello, I wanted to report that today I verified and can now assign my keyboard keystrokes to the buttons on my tablet... thanks

closed 2022-06-04 by crchmatefis