linuxwacom/xf86-input-wacom #272: "ninja -C builddir test" returns SyntaxError

Pinglinux, 2022-06-08:
@whot When running "ninja -C builddir test" with the latest code, I got:
--------cut---
1/3 xsetwacom-tests                         OK       0.01 s 
2/3 wacom-tests                             OK       0.04 s 
3/3 pytest                                  FAIL     0.37 s (exit status 4)

Ok:                    2
Expected Fail:         0
Fail:                  1
----------cut---cut---
3/3 pytest                                  FAIL     0.37 s (exit status 4)

--- command ---
01:18:26 GI_TYPELIB_PATH='/home/ping/wacom/xf86-input-wacom/builddir' LD_LIBRARY_PATH='/home/ping/wacom/xf86-input-wacom/builddir' /usr/bin/pytest
--- stderr ---
ImportError while loading conftest '/home/ping/wacom/xf86-input-wacom/test/conftest.py'.
  File "/home/ping/wacom/xf86-input-wacom/test/__init__.py", line 52
    product: int = attr.ib()
           ^
SyntaxError: invalid syntax

whot, 2022-06-08:
what python version is this? this looks like it's complaining about the type annotations but those have been a thing since python 3.5

Pinglinux, 2022-06-08:
It's on Ubuntu 20.04

$ python --version
Python 3.8.10

Checked the messages in builddir/meson-logs/testlog.txt. It has the following extra lines before the error:

--- stderr ---
No type or invalid type specified.
Must be one of stylus, touch, cursor, eraser, or pad
mod_buttons: Invalid button number 32. Insufficient storage
-------

3/3 pytest                                  FAIL     0.47 s (exit status 4)

whot, 2022-06-08:
Can you attach the `testlog.txt` please? Right now I don't really have anything to go on beyond "this fails"

Pinglinux, 2022-06-08:
Here you go: [testlog.txt](https://github.com/linuxwacom/xf86-input-wacom/files/8858116/testlog.txt)

whot, 2022-06-08:
Note the format of the output, it's divided into the various `test()` specified in `meson.build`. The stderr output you copied above is from the previous test (`wacom-tests`) and expected since it tests for this particular case.

The `pytest` error is still just the syntax error - I'm wondering now: did you pip install anything and if so did you do it as root? If not, it's likely that.

Pinglinux, 2022-06-08:
> did you pip install anything and if so did you do it as root? If not, it's likely that.

I should have used pip install before. But I don't remember if I used sudo or not. From "history | grep pip", I only got one line:

sudo apt install python3-pip

Maybe I need to do the same thing as Aaron: set up a new system?

whot, 2022-06-13:
Ok, the problem is: on ubuntu 20.04, if you install `python-pytest` it installs the python2 version of pytest only, including dragging in all of python2.7. Which was EOL 3 months before 20.04 was released...

The Python2 parser has issues with syntax added inPython3.5, unsurprisingly. The solution is to run `pytest-3` instead of `pytest`, which forces the Python3 version of the parser and then everything works:

```
host$ podman run --rm --interactive --tty --volume $PWD:/data:z ubuntu/20.04
$ cd /data
$ export DEBIAN_FRONTEND=noninteractive
$ apt-get update -y
$ apt-get install -y python3-pip jq
$ pip install yq
$ yq -r .env.UBUNTU_PACKAGES .github/workflows/build.yml | tr '\n' ' ' | xargs apt-get install -y
$ yq -r .env.PIP_PACKAGES .github/workflows/build.yml | xargs pip install
$ meson builddir -Dwacom-gobject=enabled
$ ninja -C builddir
$ pytest-3 -v --log-level=DEBUG
```

The above (admittedly typed in here so maybe there's a typo) works in podman though it skips the tests since the container won't have `/dev/uinput`. But we definitely get past that syntax error. the key is simply to invoke `pytest-3`, or alternatively, throw 20.04 out of the window and use something more up-to-date :)

The bug in meson was that it used `pytest-3` as fallback rather than as first choice - so on a system with only pytest-2 (or both) installed and where `pytest` linked to `pytest-2`, we'd pick the wrong one, causing that error. Fixed by #277

PR #277 (2022-06-13): meson: default to pytest-3 first, then pytest
At least Ubuntu 20.04 has pytest point at the Python 2.x compatible
version, resulting in syntax errors when trying to parse our Python 3
code files. So let's check pytest-3 first since that is the compatible
version and fall back to normal pytest only where it's missing.

Fixes #272
commit be347129 (2022-06-13): meson: default to pytest-3 first, then pytest

At least Ubuntu 20.04 has pytest point at the Python 2.x compatible
version, resulting in syntax errors when trying to parse our Python 3
code files. So let's check pytest-3 first since that is the compatible
version and fall back to normal pytest only where it's missing.

Fixes #272
closed 2022-06-13 by whot