linuxwacom/xf86-input-wacom #279: "ninja -C builddir test" exit status 77

Pinglinux, 2022-06-15:
@whot What did I do wrong? On Fedora 36, with the same test suite code as I tested successfully a few days ago, the 20 tests now showed in SKIP mode:

3/3 pytest          SKIP            1.09s   exit status 77
01:48:16 GI_TYPELIB_PATH=/home/ping/wacom/xf86-input-wacom-whot/builddir MALLOC_PERTURB_=128 LD_LIBRARY_PATH=/home/ping/wacom/xf86-input-wacom-whot/builddir /usr/bin/pytest
----------------------------------- output -----------------------------------
============================= test session starts ==============================
platform linux -- Python 3.10.4, pytest-6.2.5, py-1.11.0, pluggy-1.0.0
rootdir: /home/ping/wacom/xf86-input-wacom-whot
collected 20 items

test/test_wacom.py ssssssssssssssssssss                                  [100%]-------------------------------- Session errors --------------------------------
20 tests were skipped, none were run

============================= 20 skipped in 0.78s ==============================

whot, 2022-06-15:
are you running as root? The code has a [skip condition](https://github.com/linuxwacom/xf86-input-wacom/blob/master/test/__init__.py#L113) for being able to access `/dev/uinput`  and [another one](https://github.com/linuxwacom/xf86-input-wacom/blob/master/test/__init__.py#L243) for the `/dev/input/event` nodes

Pinglinux, 2022-06-15:
No, I am as myself. The only system difference between today and the other day is I updated to the latest Fedora 36. I also tried the latest xf86-input-wacom on Ubuntu 20.04. I got the same SKIP.

whot, 2022-06-15:
Unless you chmod'ed `/dev/uinput` and the event nodes, this test suite should never be able to run as root, you'd need special udev rules in place etc. Too much effort. So `sudo ninja -C builddir test` it is :)

whot, 2022-06-15:
Actually, for extra clarification and to answer the likely follow-up question: the tests skip instead of fail when they don't have access because in the glorious future we may have tests that don't require that access. In which case those tests will still run in the CI and only the ones we can't run will be skipped.

closed 2022-06-15 by whot