linuxwacom/xf86-input-wacom #284: "pan" only works if stylus in touch with tablet

fliperama00, 2022-08-05:
I have a CTL-472, and if "pan" is assigned to a button on the side of the pen, it only scrolls if the pen is in touch with the tablet, and nothing happens if I press the button while hovering the pen. Is that the intended behavior?

Pinglinux, 2022-08-09:
Did the button work without pen touching the tablet before you assigned "pan" to it?

Pinglinux, 2022-08-09:
From the [ man page](https://github.com/linuxwacom/xf86-input-wacom/blob/db26b7d0417997bd6acd06e47d97888e1a10adb4/man/xsetwacom.man#L138), "dragging" triggers the "pan" action.

fliperama00, 2022-08-09:
> Did the button work without pen touching the tablet before you assigned "pan" to it?

Yes, it did. 

I know the manual says it only works when dragging, but I was expecting for it to scroll when dragging and pan when not dragging. If it's not supposed to do that, is there a way to implement this behavior?

Pinglinux, 2022-08-10:
Is there an use case that tip touching the tablet is unaccepted ?

Pinglinux, 2022-08-10:
Ok, that's good. Let's keep the feature as it was designed, until you or someone else has time to write a patch for it.

closed 2022-08-10 by Pinglinux