linuxwacom/xf86-input-wacom #289: No button events emitted (Gnome Shell, X11, Ubuntu)

rubienr, 2022-09-10:
**Description**

If a button on the pad is configured to emit X11 mouse button events, this events are not emitted. In contrast if configured to emit key stroke events, those are emitted.

**How to reproduce**

1.  Attach device and list devices:

        xsetwacom list devices
        Wacom Intuos Pro L Pad pad      	id: 13	type: PAD
        ...

2. Define button event (unexpected behaviour): 

       xsetwacom --set 13 Button 13 button 1
       xsetwacom --get 13 Button 13         
       button +1 -1 

    Test with `xev`: no mouse event is emitted when `Button 12` is pressed on pad, whereas if the left mouse button is clicked then `button 1` events (press + release) are emitted.

**Expected behaviour**

1. Follow steps 1-2 described above.

2. Define key press event (expected behaviour):

       xsetwacom --set 13 Button 13 key x
       xsetwacom --get 13 Button 13       
       key +x -x 

    Test with `xev`: `x` key stroke events (press + release) are emitted when `Button 12` is pressed on pad.

**Environment**
````
xsetwacom --version
1.0.0

gnome-shell --version
GNOME Shell 42.4

# running X not Wayland:
echo $XDG_SESSION_TYPE
x11

X -version
X.Org X Server 1.21.1.3
X Protocol Version 11, Revision 0
Current Operating System: Linux r9 5.15.0-47-generic #51-Ubuntu SMP Thu Aug 11 07:51:15 UTC 2022 x86_64
Kernel command line: BOOT_IMAGE=/boot/vmlinuz-5.15.0-47-generic root=UUID=386f9ec2-0030-460c-b1c1-166123f2c10c ro quiet consoleblank=0 noplymouth
xorg-server 2:21.1.3-2ubuntu2.1 (For technical support please see http://www.ubuntu.com/support) 
Current version of pixman: 0.40.0
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.

cat /etc/issue        
Ubuntu 22.04.1 LTS \n \l

uname -a
Linux r9 5.15.0-47-generic #51-Ubuntu SMP Thu Aug 11 07:51:15 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
````
Reproduced with Wacom Intuos Pro L attached by 
- USB and 
- Bluetooth.

Pinglinux, 2022-09-13:
The issue is most likely caused by Gnome interfering with xsetwacom. Please take a close look at [Tablet Configuration 3: GNOME 3](https://github.com/linuxwacom/xf86-input-wacom/wiki/Tablet-Configuration-3%3A-GNOME-3) and [xsetwacom](https://github.com/linuxwacom/xf86-input-wacom/wiki/xsetwacom).

rubienr, 2022-09-17:
Thank you for the reply. I tried again to get it running with gnome-shell but i wasn't able to fix it.

What I did:
- read the links you provided in the previous post
- tried to disable `gsd-wacom`; didn't find out how to do on Ubuntu 22.04 (it is not a system service as it was in 20.04 LTS)
- brutally killed service `killall gsd-wacom`

However, my solution is to switch from gnome-shell to plasma (KDE). Unfortunately gnome-shell is still way to unstable.

Thanks for your effort!

closed 2022-09-17 by rubienr