linuxwacom/xf86-input-wacom #290: Intuos 3 update README Pad mapping

Jose-Verdu-Diaz, 2022-09-22:
I have an Intuos 3. The [README](https://github.com/linuxwacom/xf86-input-wacom/wiki/Tablet-Configuration-1%3A-xsetwacom-and-xorg.conf#intuos3) states that the valid Pad Buttons id are 1,2,3,4,5,6,7,8. On my tablet, the valid buttons are 1,2,3,8,9,10,11,12. I didn't research the reason for it, but I got it working using these new button ids.

It might be useful to add this information to the README.

Jose-Verdu-Diaz, 2022-09-22:
For reference, this is the config I used for the pad:

```bash
xsetwacom set "Wacom Intuos3 6x8 Pad pad" Button 1 "key KP_Add"
xsetwacom set "Wacom Intuos3 6x8 Pad pad" Button 2 "key KP_Subtract"
xsetwacom set "Wacom Intuos3 6x8 Pad pad" Button 3 "key 1"
xsetwacom set "Wacom Intuos3 6x8 Pad pad" Button 8 "key Prior"
xsetwacom set "Wacom Intuos3 6x8 Pad pad" Button 9 "key d"
xsetwacom set "Wacom Intuos3 6x8 Pad pad" Button 10 "key a"
xsetwacom set "Wacom Intuos3 6x8 Pad pad" Button 11 "key 4"
xsetwacom set "Wacom Intuos3 6x8 Pad pad" Button 12 "key Next"
```

Pinglinux, 2022-09-29:
Page updated. Thank you for the report!

closed 2022-09-29 by Pinglinux