linuxwacom/xf86-input-wacom #292: 3rd button of Pro Pen 3D doesn't work anymore.

enesaltinkaya, 2022-10-23:
I used to use the following commands to set my stylus buttons.
All 3 side buttons of the stylus would be mapped to mouse middle click.
The bigger button in the middle doesn't work anymore, it acts as if it is not pushed.

It's been a long while since i last used my wacom so its not clear when it stopped working.

```
xsetwacom --set "Wacom Intuos Pro M Pen stylus" Button 1 1
xsetwacom --set "Wacom Intuos Pro M Pen stylus" Button 2 2
xsetwacom --set "Wacom Intuos Pro M Pen stylus" Button 3 2
xsetwacom --set "Wacom Intuos Pro M Pen stylus" Button 8 2
```
Image of the stylus;
[image]

Pinglinux, 2022-10-28:
Did you change anything on your system? I mean did you upgrade the OS or update the driver? We need to figure out any software changes first.

jigpu, 2022-11-01:
Can confirm the issue and have bisected the problem to 6d09085b38e5ace0001cf4f3cdbd4bf247533b61 ("HID: wacom: Adding Support for new usages"). Will need to do additional study to find a solution.

enesaltinkaya, 2022-11-01:
> Did you change anything on your system? I mean did you upgrade the OS or update the driver? We need to figure out any software changes first.

Sorry for late reply, i must have missed the email when you replied.

Last time I used my wacom and pen was probably a year back, I was using arch linux back then.
Now I'm using debian, so software wise everything changed :)

jigpu, 2022-11-02:
I've just pushed a branch to my repository that fixes the issue for me. Please follow these steps to verify that it also works for you:

1. Install the `git` package
2. Run `git clone https://github.com/jigpu/input-wacom.git -b fix-292`
3. Enter the input-wacom directory that gets created
4. Follow the instructions at https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source to build and install this modified driver
5. After manually reloading the driver or rebooting, test to see if the fix works for you

enesaltinkaya, 2022-11-03:
Following your instructions fixed it for me as well.

Thanks.

jigpu, 2022-11-03:
Thanks for the confirmation. I've sent the patch up to the kernel mailinglist. If it is accepted I'll add it to the main branch of input-wacom as well.

jigpu, 2022-11-09:
Commit has been accepted and should make it into Linux 6.1, as well as be backported to stable kernels. I've imported it into the input-wacom repository as well so that it will be included in its next release.

closed 2022-11-09 by jigpu

enesaltinkaya, 2022-11-09:
Thank you for all your hard work, you guys rock.