linuxwacom/xf86-input-wacom #293: Serial Wacom Tablet WACf004 no work.

khseal, 2022-10-28:
buntu 18.04.6 LTS (GNU/Linux 5.3.5-050305-generic i686)
Log Xorg.0.log 

`[    20.759] (II) Using input driver 'wacom' for 'Serial Wacom Tablet WACf004'
[    20.759] (**) Serial Wacom Tablet WACf004: always reports core events
[    20.759] (**) Option "Device" "/dev/ttyS4"
[    20.760] (**) Option "BaudRate" "19200"
[    20.760] (**) Option "StopBits" "1"
[    20.760] (**) Option "DataBits" "8"
[    20.760] (**) Option "Parity" "None"
[    20.760] (**) Option "Vmin" "1"
[    20.760] (**) Option "Vtime" "10"
[    20.760] (**) Option "FlowControl" "Xoff"
[    20.761] (II) Serial Wacom Tablet WACf004: type not specified, assuming 'stylus'.
[    20.761] (II) Serial Wacom Tablet WACf004: other types will be automatically added.
[    21.016] (II) Serial Wacom Tablet WACf004 stylus: serial tablet id 0x90.
[    21.016] (--) Serial Wacom Tablet WACf004 stylus: using pressure threshold of 851 for button 1
[    21.016] (**) Serial Wacom Tablet WACf004 stylus: panscroll is 0
[    21.016] (**) Serial Wacom Tablet WACf004 stylus: panscroll modified to 1300
[    21.016] (--) Serial Wacom Tablet WACf004 stylus: maxX=21240 maxY=15980 maxZ=255 resX=100000 resY=100000  tilt=disabled
[    21.016] (II) Serial Wacom Tablet WACf004 stylus: hotplugging dependent devices.
[    21.017] (EE) Serial Wacom Tablet WACf004 stylus: Invalid type 'cursor' for this device.
[    21.017] (EE) Serial Wacom Tablet WACf004 stylus: Invalid type 'touch' for this device.
[    21.017] (EE) Serial Wacom Tablet WACf004 stylus: Invalid type 'pad' for this device.
[    21.017] (II) Serial Wacom Tablet WACf004 stylus: hotplugging completed.
[    21.017] (**) Option "config_info" "udev:/sys/devices/pnp0/00:02/tty/ttyS4"
[    21.017] (II) XINPUT: Adding extended input device "Serial Wacom Tablet WACf004 stylus" (type: STYLUS, id 11)
[    21.018] (**) Serial Wacom Tablet WACf004 stylus: (accel) keeping acceleration scheme 1
[    21.018] (**) Serial Wacom Tablet WACf004 stylus: (accel) acceleration profile 0
[    21.019] (**) Serial Wacom Tablet WACf004 stylus: (accel) acceleration factor: 2.000
[    21.019] (**) Serial Wacom Tablet WACf004 stylus: (accel) acceleration threshold: 4
[    21.019] (**) Option "BaudRate" "19200"
[    21.019] (**) Serial Wacom Tablet WACf004 eraser: Applying InputClass "Wacom serial class"
[    21.019] (II) Using input driver 'wacom' for 'Serial Wacom Tablet WACf004 eraser'
[    21.019] (**) Serial Wacom Tablet WACf004 eraser: always reports core events
[    21.019] (**) Option "Device" "/dev/ttyS4"
[    21.019] (**) Option "Type" "eraser"
[    21.020] (**) Option "BaudRate" "19200"
[    21.020] (**) Option "StopBits" "1"
[    21.020] (**) Option "DataBits" "8"
[    21.020] (**) Option "Parity" "None"
[    21.020] (**) Option "Vmin" "1"
[    21.020] (**) Option "Vtime" "10"
[    21.020] (**) Option "FlowControl" "Xoff"
[    21.020] (**) Serial Wacom Tablet WACf004 eraser: panscroll is 1300
[    21.020] (**) Serial Wacom Tablet WACf004 eraser: panscroll modified to 1300
[    21.020] (--) Serial Wacom Tablet WACf004 eraser: maxX=21240 maxY=15980 maxZ=255 resX=100000 resY=100000  tilt=disabled
[    21.021] (**) Option "config_info" "udev:/sys/devices/pnp0/00:02/tty/ttyS4"
[    21.021] (II) XINPUT: Adding extended input device "Serial Wacom Tablet WACf004 eraser" (type: ERASER, id 12)
[    21.034] (**) Serial Wacom Tablet WACf004 eraser: (accel) keeping acceleration scheme 1
[    21.034] (**) Serial Wacom Tablet WACf004 eraser: (accel) acceleration profile 0
[    21.034] (**) Serial Wacom Tablet WACf004 eraser: (accel) acceleration factor: 2.000
[    21.034] (**) Serial Wacom Tablet WACf004 eraser: (accel) acceleration threshold: 4
[    24.533] (II) config/udev: removing GPU device /sys/devices/pci0000:00/0000:00:02.0/drm/card0 /dev/dri/card0
[    24.533] (II) config/udev: Adding drm device (/dev/dri/card0)
[    24.533] (II) xfree86: Adding drm device (/dev/dri/card0)
[    24.533] (EE) /dev/dri/card0: failed to set DRM interface version 1.4: Permission denied
[   134.299] (WW) Serial Wacom Tablet WACf004 eraser: bad data at 4 v=9e l=9
[   134.328] (WW) Serial Wacom Tablet WACf004 eraser: bad data at 1 v=f8 l=9
[   134.359] (WW) Serial Wacom Tablet WACf004 eraser: bad data at 1 v=80 l=9
[   134.360] (WW) Serial Wacom Tablet WACf004 eraser: bad data at 4 v=cc l=9
[   134.382] (WW) Serial Wacom Tablet WACf004 eraser: bad data at 1 v=9e l=9
[   134.420] (WW) Serial Wacom Tablet WACf004 eraser: bad data at 2 v=c3 l=9
[   134.423] (WW) Serial Wacom Tablet WACf004 eraser: bad data at 2 v=e0 l=9
[   134.491] (WW) Serial Wacom Tablet WACf004 eraser: bad data at 1 v=e0 l=9
[   134.499] (WW) Serial Wacom Tablet WACf004 eraser: bad data at 1 v=e0 l=9
[   134.526] (WW) Serial Wacom Tablet WACf004 eraser: bad data at 4 v=9e l=9
[   134.564] (WW) Serial Wacom Tablet WACf004 eraser: bad data at 2 v=80 l=9
[   134.565] (WW) Serial Wacom Tablet WACf004 eraser: bad data at 1 v=9e l=9
[   134.574] (WW) Serial Wacom Tablet WACf004 eraser: bad data at 1 v=e0 l=9
[   134.620] (WW) Serial Wacom Tablet WACf004 eraser: bad data at 1 v=e0 l=9
[   134.665] (WW) Serial Wacom Tablet WACf004 eraser: bad data at 1 v=e0 l=9
[   134.680] (WW) Serial Wacom Tablet WACf004 eraser: bad data at 1 v=e0 l=9
[   134.688] (WW) Serial Wacom Tablet WACf004 eraser: bad data at 1 v=e0 l=9
[   134.737] (WW) Serial Wacom Tablet WACf004 eraser: bad data at 1 v=f8 l=9
`
Driver input-wacom in kernal work bad. Cursor jumps like crazy. In Windows all ok.
Any ideas?

Pinglinux, 2022-10-31:
How did you configure the device? Did it work on Linux before? If yes, on which distro and with which version?

khseal, 2022-10-31:
Previously worked on ubuntu version 12.04.  The device has not been configured. Worked out of the box.
When working with input-wacom, it stabilizes the movement parameter xsetwacom  --set 11 RawSample 1.

jigpu, 2022-11-01:
The built-in support for serial devices was recently removed in favor of relying on isdv4-serial-inputattach and the wacom_w8001 kernel driver. You will need to use that input-wacom driver.

Reducing RawSample to 1 should make the cursor even more jumpy. You might try installing the "evemu-tools" package and then running `sudo evemu-record` to gather some logs of the kernel events being sent. You might also try temporarily booting into an updated Ubuntu or Fedora live USB image to see if the cursor is jumpy in those environments. Ubuntu 18.04 is almost end of life...

closed 2023-03-27 by khseal