linuxwacom/xf86-input-wacom #295: Touch Off, Palm (multitouch) Freezes Stylus

shawnp0wers, 2022-11-08:
I'm running Debian Bookworm with my Cintiq Pro 13. I'm using xsetwacom to configure the tablet, and it's 99% perfect. I have two issues, one is annoying, but one is hard to work around.

Easy issue:
When my computer monitor turns back on after blanking (xscreensaver), the wacom settings are back to default, so I need to rerun the xsetwacom to disable touch and map the stylus/eraser to my wacom screen. Not a big deal, I just run a quick script.

Difficult challenge:
Even with touch turned off, when I'm drawing, my palm seems to trigger "something" that makes my stylus unresponsive until I lift my palm and touch the screen with my finger. Specifically, I need to touch the tablet in two separate locations (two fingers an inch apart will do the same as my palm), and I have to hold them in place for 2-3 seconds. Then the stylus acts like it doesn't exist. But if I lift my palm (or two fingers) and touch the screen with a single finger, the stylus starts working again.

Again, touch is turned off, and there doesn't appear to be any OS interaction with my palm/fingers touching the screen. It just seems to disable the stylus until I do that "tap with one finger" thing. I've actually ordered a palm rejection glove to wear, but that's a kludgy workaround. I don't know what is actually happening with the tablet. I've tried fiddling with every option on all devices (pad, touch, stylus, etc) -- any insight, or even a method to see if the tablet is reporting "something" would be greatly appreciated.

Pinglinux, 2022-11-13:
Before any troubleshooting steps, I'd like to know if you are running the wacom.ko built from the latest input-wacom and wacom_drv.so from the latest xf86-input-wacom.

shawnp0wers, 2022-11-13:
I haven't done any compilation myself -- I'm using Debian Testing (bookworm), with the Debian 6.0.5-1 kernel.

> apt info xserver-xorg-input-wacom
> Package: xserver-xorg-input-wacom
> Version: 1.1.0-1

The commands I run when starting Xwindows to "set up" my tablet are:

> xsetwacom --set "Wacom Cintiq Pro 13 Pen stylus" MapToOutput HEAD-2
> xsetwacom --set "Wacom Cintiq Pro 13 Pen eraser" MapToOutput HEAD-2
> xsetwacom --set "Wacom Cintiq Pro 13 Touch Finger touch" touch off

If I can provide any other info, please let me know. And thank you in advance for any thoughts.

Here is a demonstration of the odd behavior: https://youtu.be/HyTa91ZhWhY

Pinglinux, 2022-11-16:
I tested a pen and touch enabled tablet (I don't have a Cintiq Pro 13 around at the moment) on Fedora 36 running kernel 6.0.8. I could not reproduce the issue. Another detail of my system is it's under Gnome in Xorg desktop environment (DE). What DE do you use?

With that said, please post the output of "xinput list", dmesg, and Xorg.0.log (you may need to search for the directory in which the log file is stored since different distro could store it differently. On my Fedora 36, it is under ~/.local/share/xorg). I'll follow up with you after we have those details.

Thank you for making the youtube video. That's helpful.

shawnp0wers, 2022-11-16:
I appreciate the continued help!

I'm not using a Desktop Manager/Environment, just Xorg started with startx. I have a compositor (picom) in my .xsession file along with xscreensaver and then I launch icewm-session. (I had weeks of issues getting a DE to work with my multiple screens of varying scale and orientation)

xinput list gives:

```
spowers@bigtuna:~$ xinput list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Logitech M325                           	id=8	[slave  pointer  (2)]
⎜   ↳ Logitech M600                           	id=9	[slave  pointer  (2)]
⎜   ↳ Logitech K400 Plus                      	id=10	[slave  pointer  (2)]
⎜   ↳ P. I. Engineering XK-16 HID             	id=15	[slave  pointer  (2)]
⎜   ↳ Wacom Cintiq Pro 13 Touch Finger touch  	id=16	[slave  pointer  (2)]
⎜   ↳ Wacom Cintiq Pro 13 Pad pad             	id=17	[slave  pointer  (2)]
⎜   ↳ Wacom Cintiq Pro 13 Pen stylus          	id=18	[slave  pointer  (2)]
⎜   ↳ Wacom Cintiq Pro 13 Pen eraser          	id=19	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Power Button                            	id=7	[slave  keyboard (3)]
    ↳ Logitech K400 Plus                      	id=14	[slave  keyboard (3)]
    ↳ P. I. Engineering XK-16 HID             	id=11	[slave  keyboard (3)]
    ↳ HD Pro Webcam C920                      	id=13	[slave  keyboard (3)]
```

My dmesg output  is constant complaint about my HDMI to USB capture device, and nothing else. Here is the snippet that repeats:

```
[23838.436691] usb 8-2: USB disconnect, device number 103
[23839.063033] usb 8-2: new SuperSpeed USB device number 104 using xhci_hcd
[23839.088327] usb 8-2: New USB device found, idVendor=1e4e, idProduct=701f, bcdDevice= 1.00
[23839.088332] usb 8-2: New USB device strings: Mfr=6, Product=7, SerialNumber=3
[23839.088335] usb 8-2: Product: HDMI to U3 capture
[23839.088336] usb 8-2: Manufacturer: Video Grabber
[23839.088338] usb 8-2: SerialNumber: 20000130041415
[23839.088722] usb 8-2: Found UVC 1.00 device HDMI to U3 capture (1e4e:701f)
[23839.090961] usb 8-2: UVC non compliance - GET_DEF(PROBE) not supported. Enabling workaround.
[23850.127032] usb 8-2: reset SuperSpeed USB device number 104 using xhci_hcd
[23851.382370] usb 8-2: USB disconnect, device number 104
[23852.010910] usb 8-2: new SuperSpeed USB device number 105 using xhci_hcd
[23852.036366] usb 8-2: New USB device found, idVendor=1e4e, idProduct=701f, bcdDevice= 1.00
[23852.036371] usb 8-2: New USB device strings: Mfr=6, Product=7, SerialNumber=3
[23852.036374] usb 8-2: Product: HDMI to U3 capture
[23852.036376] usb 8-2: Manufacturer: Video Grabber
[23852.036378] usb 8-2: SerialNumber: 20000130041415
[23852.036759] usb 8-2: Found UVC 1.00 device HDMI to U3 capture (1e4e:701f)
[23852.037113] usb 8-2: UVC non compliance - GET_DEF(PROBE) not supported. Enabling workaround.
[23860.119161] usb 8-2: Device not responding to setup address.
[23860.327040] usb 8-2: Device not responding to setup address.
[23860.534886] usb 8-2: device not accepting address 105, error -71
[23860.738913] usb 8-2: reset SuperSpeed USB device number 105 using xhci_hcd
[23862.092614] usb 8-2: USB disconnect, device number 105
[23862.722909] usb 8-2: new SuperSpeed USB device number 106 using xhci_hcd
```
The Xorg.0.log file was in my /var/log folder, but is quite long, so I put its contents here:
https://pastebin.com/06fq4qw6

Thank you again, and if I've gathered the wrong info, just let me know. :)

shawnp0wers, 2022-11-16:
Also, while the Xorg log file shows, I'm using an NVIDIA card. In order to set up my screens, I execute this as part of my .xsession before picom or icewm:

```
nvidia-settings --assign "CurrentMetaMode=\
DPY-0: nvidia-auto-select @1440x2560 +0+0 {AllowGSYNC=Off,ViewPortIn=1440x2560, ViewPortOut=2560x1440+0+0, Rotation=90}, \
DPY-5: nvidia-auto-select @3072x1728 +1440+300 {Transform=(0.799988,0.000000,0.000000,0.000000,0.799988,0.000000,0.000000,0.000000,1.000000), AllowGSYNC=Off, ViewPortIn=3072x1728, ViewPortOut=3840x2160+0+0, ResamplingMethod=Bilinear}, \
DPY-1: nvidia-auto-select @1920x1080 +1910+2027 {AllowGSYNC=Off, ViewPortIn=1920x1080, ViewPortOut=1920x1080+0+0}, \
DPY-3: nvidia-auto-select @3072x1728 +4511+300 {Transform=(0.799988,0.000000,0.000000,0.000000,0.799988,0.000000,0.000000,0.000000,1.000000), AllowGSYNC=Off, ViewPortIn=3072x1728, ViewPortOut=3840x2160+0+0, ResamplingMethod=Bilinear}"

nvidia-settings -a XineramaInfoOrder=DP-4
```

Pinglinux, 2022-11-17:
@shawnp0wers Thanks for the details. I was thinking there may be another touch enabled device interfering with the tablet. It looks like that's not the case. Can you do two more testing for me.

1. Issue a {xsetwacom get "Wacom Cintiq Pro 13 Touch Finger touch" touch} command before reproducing the issue. You should see "off" or "0" from the output. If not, let me know. Otherwise, issue the same command after you see the issue;

2. Turn touch off in xorg.conf instead of by xsetwacom to see if it makes a difference or not.

shawnp0wers, 2022-11-17:
It showed "off" before the issue, I got it to "do the thing" and while it was unresponsive to the stylus I ran the command again, and it still showed off. Tapping the tablet (to end whatever "the thing" is) and checking still showed off.

I'm not sure how to configure the wacom stuff in xorg.conf, I'll have to do some googling to figure that out, but I'll update here once I get it. :)

Thanks again,
-Shawn

shawnp0wers, 2022-11-18:
This is proving remarkably challenging for me. After work, I'll shut down my X server and disable nodm so that I can generate an xorg.conf file which I'll hopefully be able to turn into an appropriate 50-wacom.conf file. I'm firing neurons which rusted solid back in the early 2000s... :)

-Shawn

Pinglinux, 2022-11-19:
@shawnp0wers If that's too much trouble, we could turn the X driver debug on to see if there are any touch events processed in the driver. Please take a look at https://github.com/linuxwacom/xf86-input-wacom/wiki/Debugging and play with the debug levels to get a sense of the touch events.

Pinglinux, 2023-09-29:
We can't figure out the root cause without further details.

closed 2023-09-29 by Pinglinux