linuxwacom/xf86-input-wacom #300: Wacom One Small is not recognized as compatible

Mart-Bogdan, 2022-12-13:
When I run xsetwacom

```
 xsetwacom -v --list devices                                                                                                                                                                                                 
... Display is '(null)'.
... 'list' requested.
...........
... Found device 'Wacom One by Wacom S Pen' (22).
... 'Wacom One by Wacom S Pen' (22) is not a wacom device.
... Found device 'Wacom One by Wacom S Pen Pen (0)' (23).
... 'Wacom One by Wacom S Pen Pen (0)' (23) is not a wacom device.
```
Non wacom USB devices trimmed from output for brevity.
varsion:
```
 xsetwacom -V                
1.1.0
```

Device is https://rozetka.com.ua/wacom_ctl_472_n/p24751209/

But device is detected by kernel and by Xorg. So I can use it in Krita(for example), but can't configure it. Neither using **xsetwacom** nor wirh KDE utility. It doesn't see device.

Kernel: 5.15.81-1-MANJARO
DM:  KDE Plasma, Xorg

whot, 2022-12-13:
Those device names suggest you're using the xf86-input-libinput driver, not the wacom driver.  Please make sure the wacom driver is installed and assigned to the tablet devices.

Mart-Bogdan, 2022-12-23:
Wow. Interesting. didn't know about that. Thanks, @whot, for pointing it out. I'll check what can be done.

whot, 2023-01-08:
Closing this for now, please re-open if it still doesn't work with the wacom driver, thanks.

closed 2023-01-08 by whot