linuxwacom/xf86-input-wacom #301: Wacom Intuos with strange stylus behaviour

ChrisDAT20, 2022-12-26:
Got a Wacom Intuos tablet with a pen which has no eraser. 
The device is detected automatically and can be used immediately after plugging in (Linux Mint Debian Etsie, laptop with only one screen, X server on onboard Intel, Nvidia device used only for ML).
`xsetwacom`reports:
Wacom Co.,Ltd. Intuos S stylus  	id: 21	type: STYLUS    
Wacom Co.,Ltd. Intuos S eraser  	id: 22	type: ERASER

As I am new to tablets, I had a while to practise till I could use the pen "somehow".
I watched some tutorial videos on how to use the device with Gimp or Krita and yes, everything looks so easy.
I played around with absolute and relative mode, and changed the Area and MapToOutput parameters.
```
Option "Area" "0 0 200 112"
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Option "DebugLevel" "0"
Option "CommonDBG" "0"
Option "Suppress" "2"
Option "RawSample" "4"
Option "PressCurve" "0 0 100 100"
Option "Mode" "Relative"
Option "TPCButton" "off"
Option "Touch" "off"
Property 'Wacom Hardware Touch Switch' does not exist on device.
Option "Gesture" "off"
Option "ZoomDistance" "0"
Option "ScrollDistance" "0"
Option "TapTime" "250"
Property 'Wacom Proximity Threshold' does not exist on device.
Option "Rotate" "none"
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Option "Threshold" "26"
Option "Serial" "0"
Option "PressureRecalibration" "on"
```
However, I can move around with the pen and the input pointer on screen follows the movements.
BUT: the input pointer on screen jumps around quite frequently. Not only by a few pixels, its sometimes jumps to the opposite side (up/down, left/right and vice versa). This at a distance of the pen from the tablet of around 1-2cm.
And when I lower the pen towards the tablet, the input pointer jumps around heavily, it is almost impossible to use the pen for some input or drawing. And it seems that the tip of the pen is already pressed so that actions are triggered on the app which is below the input pointer.

Do I need to adjust some other settings as well to be able to use the pen properly and with accuracy?

whot, 2023-01-05:
This could be a hardware issue. Can you please record your device with `libinput record` (see [here](https://wayland.freedesktop.org/libinput/doc/latest/tools.html#libinput-record-and-libinput-replay)) and *attach* the recording here. This should immediately show if the jumps are generated in the device or in the driver or elsewhere.

Pinglinux, 2023-09-27:
I see you turned Option "Mode" "Relative". If you want to draw/write something, you should stay at Absolute mode. Otherwise the cursor would jump like a mouse.

Also, you should not set the tablet area to Option "Area" "0 0 200 112". You want to use the whole tablet area since mapping a smaller tablet area to the whole screen/output would make the cursor move fast too.

You probably have already figured that out by now.

closed 2023-09-27 by Pinglinux