linuxwacom/xf86-input-wacom #303: Relative mode has cursor acceleration

maeries, 2023-01-08:
When in the Gnome settings I turn off tablet mode (meaning switch to mouse mode aka relative mode) I noticed that the cursor has now some acceleration to it. It's like mouse acceleration that you can set in Windows and Linux. Meaning, if you move the pen faster, it will move the cursor further, ever if the distance the pen travels is the same. It can be easily checked by placing a ruler on your tablet horizontally and the tip of you pen on its edge. Now move the pen up slowly and then quickly down again against the ruler. The cursor will now be lower that it was before.

Is there a way to disable that? It makes writing in relative mode look like I'm drunk. I have already disabled mouse acceleration.

Setup:

- Manjaro with Gnome 43.2
- X11
- Intuos M Bluetooth

Pinglinux, 2023-09-29:
You don't want to write or draw in relative mode. You lost the accuracy that pen offers you. Relative mode is to help you reach out to/move something on the other side of the screen. You switch mode between those two activities with the same pen.

jigpu, 2023-10-13:
How did you disable mouse acceleration? Acceleration is controlled per-device, so you will need to disable it on the stylus as well. This is not an option that GNOME exposes, but can be done through the command-line "xinput" tool. For example, running the following command should disable acceleration on the device with the given name:

~~~
$ xinput set-prop "<name>" "Device Accel Profile" -1
~~~

If you replace <name> with the name of your stylus, it should (temporarily) disable acceleration on it in relative mode. Note that you will need to re-run this command any time you log in or unplug/replug your device.

closed 2023-10-13 by jigpu