linuxwacom/xf86-input-wacom #305: `autogen.sh` is creating paths with a literal "$(top_srcdir)"

andreareina, 2023-02-03:
`autogen.sh` is creating paths with a literal "$(top_srcdir)", with the result that make can't find the files:
```
Makefile:530: ../src/.deps/wacom_drv_la-wcmCommon.Plo: No such file or directory
```

```
~/src/xf86-input-wacom-1.1.0 $ sed -n 530p src/Makefile
include $(top_srcdir)/src/$(DEPDIR)/wacom_drv_la-wcmCommon.Plo
```

OS: Ubuntu 18.04 LTS (yeah I've been lazy about upgrading :sweat_smile:)

Terminal session:
```
/tmp $ tar -xf ~/Downloads/xf86-input-wacom-1.1.0.tar.bz2 
0 ✓ (99.6ms) 18:41:06 andrea@andrea-desktop 
/tmp $ cd xf86-input-wacom-1.1.0/
0 ✓ (6.4ms) 18:41:11 andrea@andrea-desktop 
/tmp/xf86-input-wacom-1.1.0 $ find . -name *Plo
0 ✓ (9.6ms) 18:41:16 andrea@andrea-desktop 
/tmp/xf86-input-wacom-1.1.0 $ ./autogen.sh 
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal 
autoreconf: configure.ac: tracing
autoreconf: running: libtoolize --copy
libtoolize:   error: './ltmain.sh' is newer: use '--force' to overwrite
libtoolize:   error: 'm4/libtool.m4' is newer: use '--force' to overwrite
libtoolize:   error: 'm4/ltversion.m4' is newer: use '--force' to overwrite
libtoolize: Consider adding '-I m4' to ACLOCAL_AMFLAGS in Makefile.am.
autoreconf: running: /usr/bin/autoconf
autoreconf: running: /usr/bin/autoheader
autoreconf: running: automake --add-missing --copy --no-force
autoreconf: Leaving directory `.'
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... yes
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking minix/config.h usability... no
checking minix/config.h presence... no
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
[... 127 lines cut ...]
config.status: creating test/Makefile
config.status: creating xorg-wacom.pc
config.status: creating config.h
config.status: executing depfiles commands
config.status: executing libtool commands
0 ✓ (10.9s) 18:41:31 andrea@andrea-desktop 
/tmp/xf86-input-wacom-1.1.0 $ find . -name *Plo
./src/$(top_srcdir)/src/x11/.deps/wacom_drv_la-xf86WacomProperties.Plo
./src/$(top_srcdir)/src/x11/.deps/wacom_drv_la-xf86Wacom.Plo
./src/$(top_srcdir)/src/.deps/wacom_drv_la-wcmConfig.Plo
./src/$(top_srcdir)/src/.deps/wacom_drv_la-wcmTouchFilter.Plo
./src/$(top_srcdir)/src/.deps/wacom_drv_la-wcmFilter.Plo
./src/$(top_srcdir)/src/.deps/wacom_drv_la-wcmValidateDevice.Plo
./src/$(top_srcdir)/src/.deps/wacom_drv_la-wcmCommon.Plo
./src/$(top_srcdir)/src/.deps/wacom_drv_la-wcmUSB.Plo
./test/.deps/wacom_drv_test_la-wacom-test-suite.Plo
./test/$(top_srcdir)/src/x11/.deps/wacom_drv_test_la-xf86WacomProperties.Plo
./test/$(top_srcdir)/src/x11/.deps/wacom_drv_test_la-xf86Wacom.Plo
./test/$(top_srcdir)/src/.deps/wacom_drv_test_la-wcmCommon.Plo
./test/$(top_srcdir)/src/.deps/wacom_drv_test_la-wcmUSB.Plo
./test/$(top_srcdir)/src/.deps/wacom_drv_test_la-wcmValidateDevice.Plo
./test/$(top_srcdir)/src/.deps/wacom_drv_test_la-wcmFilter.Plo
./test/$(top_srcdir)/src/.deps/wacom_drv_test_la-wcmTouchFilter.Plo
./test/$(top_srcdir)/src/.deps/wacom_drv_test_la-wcmConfig.Plo
```

andreareina, 2023-02-03:
Same thing happens using the exact invocation given in the wiki

```
/tmp $ rm -rf xf86-input-wacom-1.1.0/
0 ✓ (11.4ms) 18:48:55 andrea@andrea-desktop 
/tmp $ tar xf ~/Downloads/xf86-input-wacom-1.1.0.tar.bz2 
0 ✓ (127ms) 18:49:02 andrea@andrea-desktop 
/tmp $ cd xf86-input-wacom-1.1.0/
0 ✓ (6.8ms) 18:49:04 andrea@andrea-desktop 
/tmp/xf86-input-wacom-1.1.0 $ set -- --prefix="/usr" --libdir="$(readlink -e $(ls -d /usr/lib*/xorg/modules/input/../../../ | head -n1))"
0 ✓ (19.4ms) 18:49:08 andrea@andrea-desktop 
/tmp/xf86-input-wacom-1.1.0 $ if test -x ./autogen.sh; then ./autogen.sh "$@"; else ./configure "$@"; fi && make && sudo make install || echo "Build Failed"
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal 
autoreconf: configure.ac: tracing
autoreconf: running: libtoolize --copy
[...]
make[2]: Entering directory '/tmp/xf86-input-wacom-1.1.0/src'
Makefile:530: ../src/.deps/wacom_drv_la-wcmCommon.Plo: No such file or directory
Makefile:531: ../src/.deps/wacom_drv_la-wcmConfig.Plo: No such file or directory
Makefile:532: ../src/.deps/wacom_drv_la-wcmFilter.Plo: No such file or directory
Makefile:533: ../src/.deps/wacom_drv_la-wcmTouchFilter.Plo: No such file or directory
Makefile:534: ../src/.deps/wacom_drv_la-wcmUSB.Plo: No such file or directory
Makefile:535: ../src/.deps/wacom_drv_la-wcmValidateDevice.Plo: No such file or directory
Makefile:536: ../src/x11/.deps/wacom_drv_la-xf86Wacom.Plo: No such file or directory
Makefile:537: ../src/x11/.deps/wacom_drv_la-xf86WacomProperties.Plo: No such file or directory
make[2]: *** No rule to make target '../src/x11/.deps/wacom_drv_la-xf86WacomProperties.Plo'.  Stop.
make[2]: Leaving directory '/tmp/xf86-input-wacom-1.1.0/src'
Makefile:526: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/tmp/xf86-input-wacom-1.1.0'
Makefile:435: recipe for target 'all' failed
make: *** [all] Error 2
Build Failed
```

whot, 2023-02-03:
confirmed on 18.04 in a podman container. Looks like this is [an automake bug](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=752993) that has been fixed since and it no works. *possible* that the `subdir-objects` which changed at some point in the recent-ish past triggers this but... dunno.

Not sure how to fix this but I'll take any suggestions. You might have better luck installing meson and ninja from pip and installing it that way if automake can't be updated.

closed 2023-02-03 by andreareina