linuxwacom/xf86-input-wacom #306: 2fg-right-click Zoom Scroll mixed up

dodisoft, 2023-03-26:
Hello, 
After recovering the right-click feature on my Bamboo-fun last summer with help of this precious community, I finally find time to confirm a new annoyance.
From that day, as I operate a 2fg-right-click , it's very often mistaken for a Zoom or Scroll gesture. This occurs randomly and in any software (Firefox, Pcmanfm...).
I fiddled around various options in Xsetwacom, but none cured this behaviour.
As it seems, no further version than xserver-xorg-input-wacom 0.34.99.1-1 (i.e. from Debian Bullseye), works flawlessly for my tablet.

My setup :

Distro : Debian bookworm 
Kernel : 6.1.0-6-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.15-1 (2023-03-05) x86_64 GNU/Linux
Desktop envt : LXDE
Tablet : ID 056a:00d3 Wacom Co., Ltd CTH-661 [Bamboo Fun/Comic Pen & Touch (M)]
Driver : Compiled with Pinglinux patch as described here:
[https://github.com/linuxwacom/xf86-input-wacom/issues/103#issuecomment-1229610024](url)

Thanks for your work !

Dodic

Pinglinux, 2023-06-14:
Give us some time to troubleshoot the root cause...

Pinglinux, 2024-05-18:
When added Multi-Touch (MT) gestures to xsetwacom 13+ years ago, I only considered it as a workaround since xinput didn't support MT. We turned gestures off by default when MT support was stabilized in xinput2. There are many free MT solutions on Linux nowadays. Considering X server is in maintenance mode, we decided to drop this workaround and concentrate on future features.

closed 2024-05-18 by Pinglinux