linuxwacom/xf86-input-wacom #310: Wacom Intuos PT S 2 wrong button mapping

Pat4yczek, 2023-06-04:
Hi everyone.
I have problem with buttons on tablet, there are wrong mapped
can't do printscreen when buttons mapping is entered, so can show that way:

`--1-----2--`
`--3-----4--`

here you have correct input for buttons, below I have that way, where 0 mean not input:
`--3-----1--`
`--0-----4--`

Pinglinux, 2023-06-14:
What kernel version and distribution do you use? I assume your system is based on X server. How did you get those button values?

Pat4yczek, 2023-07-03:
System:
  Kernel: 5.15.0-73-generic x86_64 bits: 64 compiler: gcc v: 11.3.0 Desktop: Cinnamon 5.6.8
    tk: GTK 3.24.33 wm: muffin dm: LightDM Distro: Linux Mint 21.1 Vera base: Ubuntu 22.04 jammy

I didn't know where to look so I wrote some visualization

After I read your comment I'll go search in documentation of xsetwacom, found solution and fix it :smile: 

I used this syntax:
`xsetwacom set device_name Button button-number [mapping]`

So I have Button 1, 3, 8 and 9, my commands was:
`xsetwacom set 24 Button 1 "button +3"`
`xsetwacom set 24 Button 3 "button +1"`
`xsetwacom set 24 Button 8 "button +8"`
`xsetwacom set 24 Button 9 "button +2"`

Pat4yczek, 2023-07-03:
Now I have problem when tablet is plug out and plug in again it not save this settings...

Pinglinux, 2023-09-26:
Did you try the suggestion here: https://github.com/linuxwacom/xf86-input-wacom/wiki/xsetwacom-example-scripts?

closed 2023-09-26 by Pinglinux