linuxwacom/xf86-input-wacom #312: Wacom Intuos PT S 2 is not detected by the Wacom Tablet manager - Ubuntu 23.04

PandLantis, 2023-06-12:
I bought a new (old) Intuos Art Small that works flawlessly except that the tablet is not detected by the Wacom Tablet settings and I am not able to set the buttons on the tablet. Size mapping and pen works perfectly when it was used with GIMP, Inkscape and Krita.

I am using Ubuntu 23.04 Unity (6.2.0-20_Generic) on a Dell Precision M4700
Intel i7-3940XM
Nvidia GTX 965M
24GB DDR3

whot, 2023-06-12:
Can you please [check if libwacom shows your device](https://github.com/linuxwacom/libwacom/wiki/Check-device-support)?

PandLantis, 2023-06-13:
The definition file does match the USB vid and pid values and I used evtest to ensure that my pad was not suffering from a hardware failure.

```
> sudo evtest /dev/input/event22
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x56a product 0x33c version 0x110
Input device name: "Wacom Intuos PT S 2 Pad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 272 (BTN_LEFT)
    Event code 273 (BTN_RIGHT)
    Event code 277 (BTN_FORWARD)
    Event code 278 (BTN_BACK)
    Event code 331 (BTN_STYLUS)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max        1
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max        1
Properties:
Testing ... (interrupt to exit)
Event: time 1686616815.404231, type 1 (EV_KEY), code 273 (BTN_RIGHT), value 1
Event: time 1686616815.404231, -------------- SYN_REPORT ------------
Event: time 1686616815.540226, type 1 (EV_KEY), code 273 (BTN_RIGHT), value 0
Event: time 1686616815.540226, -------------- SYN_REPORT ------------
Event: time 1686616815.930161, type 1 (EV_KEY), code 277 (BTN_FORWARD), value 1
Event: time 1686616815.930161, -------------- SYN_REPORT ------------
Event: time 1686616816.026170, type 1 (EV_KEY), code 277 (BTN_FORWARD), value 0
Event: time 1686616816.026170, -------------- SYN_REPORT ------------
Event: time 1686616816.372238, type 1 (EV_KEY), code 272 (BTN_LEFT), value 1
Event: time 1686616816.372238, -------------- SYN_REPORT ------------
Event: time 1686616816.506239, type 1 (EV_KEY), code 272 (BTN_LEFT), value 0
Event: time 1686616816.506239, -------------- SYN_REPORT ------------
Event: time 1686616816.860243, type 1 (EV_KEY), code 278 (BTN_BACK), value 1
Event: time 1686616816.860243, -------------- SYN_REPORT ------------
Event: time 1686616816.956239, type 1 (EV_KEY), code 278 (BTN_BACK), value 0
Event: time 1686616816.956239, -------------- SYN_REPORT ------------

```

whot, 2023-06-13:
in that case the libwacom bits are working as expected, so the bug is more likely to be in gnome, either in mutter or the control center. Any chance you can try under wayland? it it works there it could still be an X driver problem (or the different backend).

PandLantis, 2023-06-13:
I will give it a try and report back.

closed 2024-05-17 by Pinglinux