linuxwacom/xf86-input-wacom #313: Left click not recognized on dell latitude 7200

Kyuhae, 2023-06-25:
Hello,

Under Ubuntu 16.04, my touchscreen detects taps using two fingers, but not with one finger, so i cannot perform left click actions.
Also, if I do a two-finger tap, I subsequently cannot perform left click actions in certain applications, with my **mouse**.
It works fine under windows 10.

For context, I previously had messed around with drivers and configuration when troubleshooting the touchscreen which turned out to be physically broken. That broken touchscreen was replaced, and this new screen works properly under windows and in the BIOS, but clearly the changes I made previously are still impacting my ubuntu 16.04 install's ability to properly use the touchscreen.

I have tried reinstalling the `xserver-xorg-input-wacom` driver, which made no changes.

I tried searching for information, but I am not knowledgeable at all in the area of Xorg configuration / xinput, and so on, and all the leads I have followed usually took me to trying to figure out which events correspond to which buttons, at which point I am lost.

I know this screen worked with this version of ubuntu, so it seems like there should be some way to wipe all changes and revert back to defaults. Could anyone guide me trough the process?

I've run the sysinfo script to give you as many details as possible.

Many thanks for any help you might be able to give! 

[sysinfo.3c63ZJFlWo.tar.gz](https://github.com/linuxwacom/xf86-input-wacom/files/11860161/sysinfo.3c63ZJFlWo.tar.gz)

whot, 2023-06-25:
>  Also, if I do a two-finger tap, I subsequently cannot perform left click actions in certain applications, with my mouse.

this usually indicates a "stuck button" where the previous tap doesn't release the button correctly, leading to it being permanently down.

> which events correspond to which buttons, at which point I am lost.

`xinput test-xi2` will show you the various button events that an X client sees. The `libinput record` tool ([docs](https://wayland.freedesktop.org/libinput/doc/latest/tools.html)) will show the events coming out of the kernel, that's usually a good way to check if the device behaves correctly.

Mind you - 16.04 is now 7y old so there's a good chance that bug was fixed already - upgrading or at least testing a more recent version will be a good idea.

Kyuhae, 2023-06-27:
Thanks for your reply!
I'm just way out of my depth, so unless someone can help me if i post the output of such commands, i don't think i'll be able to do much. Also while trying to install `libinput`, aptitude suggests removing `xserver-xorg` which I will definitely not do (i assume that's a symptom of something else being broken in my install).

Regarding upgrading, like I said I know it worked even with this version of ubuntu, and still works with live usb with various other versions. I'm tied to 16.04 because of legacy code and installs that require it for my research work, and similarly would like to avoid having to reinstall 16.04. The fact that it worked with 16.04 before is why I'm wondering if there is a simple way to essentially re-initialize everything related to the touchscreen back to system defaults.

whot, 2023-07-03:
>  Also while trying to install libinput, aptitude suggests removing xserver-xorg which I will definitely not do (i assume that's a symptom of something else being broken in my install).

uhm, yes, definitely. something isn't right there, libinput is just a library that can sit there without being used, installing it should *not* remove the server.

Buttons are mostly handled in the server, so maybe there was a recent update to the X server that caused this? I don't remember the rest of the stack that 16.04 so I'm not sure where else to point the finger to.

All the tablet-specific configuration will be in /etc/X11/xorg.conf.d or /usr/share/X11/xorg.conf.d but unless you manually put a file in there that should just be the default driver assignments anyway. So, unfortunately ... :shrug:  no idea

Pinglinux, 2024-05-17:
Please reopen the issue if you still face it.

closed 2024-05-17 by Pinglinux