linuxwacom/xf86-input-wacom #314: Wacom CTL-672 not working on login

mod20388, 2023-07-07:
Wacom CTL-672 is not functional on login. It is only responsive after `rmmod wacom && modprobe wacom` (possibly a couple times).

Relevant lines in Xorg.0.log appear, so the tablet is recognized during boot but nevertheless not functional:
`[    18.268] (II) config/udev: Adding input device Wacom One by Wacom M Pen (/dev/input/event3)
[    18.268] (**) Wacom One by Wacom M Pen: Applying InputClass "libinput tablet catchall"
[    18.268] (**) Wacom One by Wacom M Pen: Applying InputClass "Wacom USB tablet class"
[    18.268] (**) Wacom One by Wacom M Pen: Applying InputClass "Wacom tablet class"
[    18.268] (II) LoadModule: "wacom"
[    18.268] (II) Loading /usr/lib/xorg/modules/input/wacom_drv.so
[    18.269] (II) Module wacom: vendor="X.Org Foundation"
[    18.269] (II) wacom: Driver for Wacom graphics tablets: PenPartner, Graphire,
        Wacom One Pen Display 13, Intuos2 4x5, Intuos2 6x8, Intuos2 9x12,
[    18.271] (II) Using input driver 'wacom' for 'Wacom One by Wacom M Pen'
[    18.271] (**) Wacom One by Wacom M Pen: always reports core events
[    18.326] (II) wacom: Wacom One by Wacom M Pen: type not specified, assuming 'stylus'.
[    18.326] (II) wacom: Wacom One by Wacom M Pen: other types will be automatically added.
[    18.326] (--) wacom: Wacom One by Wacom M Pen stylus: using pressure threshold of 851 for button 1
[    18.326] (**) wacom: Wacom One by Wacom M Pen stylus: panscroll is 0
[    18.326] (**) wacom: Wacom One by Wacom M Pen stylus: panscroll modified to 1300
[    18.326] (--) wacom: Wacom One by Wacom M Pen stylus: maxX=21600 maxY=13500 maxZ=2047 resX=100000 resY=100000  tilt=enabled
[    18.326] (II) wacom: Wacom One by Wacom M Pen stylus: hotplugging dependent devices.
[    18.326] (II) wacom: Wacom One by Wacom M Pen stylus: hotplugging completed.
[    18.400] (II) XINPUT: Adding extended input device "Wacom One by Wacom M Pen stylus" (type: STYLUS, id 9)
[    18.400] (**) Wacom One by Wacom M Pen stylus: (accel) keeping acceleration scheme 1
[    18.400] (**) Wacom One by Wacom M Pen stylus: (accel) acceleration profile 0
[    18.400] (**) Wacom One by Wacom M Pen stylus: (accel) acceleration factor: 2.000
[    18.400] (**) Wacom One by Wacom M Pen stylus: (accel) acceleration threshold: 4
[    18.457] (II) config/udev: Adding input device Wacom One by Wacom M Pen (/dev/input/mouse1)
[    18.697] (**) Wacom One by Wacom M Pen eraser: Applying InputClass "libinput tablet catchall"
[    18.697] (**) Wacom One by Wacom M Pen eraser: Applying InputClass "Wacom USB tablet class"
[    18.697] (**) Wacom One by Wacom M Pen eraser: Applying InputClass "Wacom tablet class"
[    18.697] (II) Using input driver 'wacom' for 'Wacom One by Wacom M Pen eraser'
[    18.697] (**) Wacom One by Wacom M Pen eraser: always reports core events
[    18.697] (**) wacom: Wacom One by Wacom M Pen eraser: panscroll is 1300
[    18.697] (**) wacom: Wacom One by Wacom M Pen eraser: panscroll modified to 1300
[    18.697] (--) wacom: Wacom One by Wacom M Pen eraser: maxX=21600 maxY=13500 maxZ=2047 resX=100000 resY=100000  tilt=enabled
[    18.736] (II) XINPUT: Adding extended input device "Wacom One by Wacom M Pen eraser" (type: ERASER, id 15)
[    18.736] (**) Wacom One by Wacom M Pen eraser: (accel) keeping acceleration scheme 1
[    18.736] (**) Wacom One by Wacom M Pen eraser: (accel) acceleration profile 0
[    18.736] (**) Wacom One by Wacom M Pen eraser: (accel) acceleration factor: 2.000
[    18.736] (**) Wacom One by Wacom M Pen eraser: (accel) acceleration threshold: 4`

Pinglinux, 2023-09-26:
Can you tell us the Linux distro and its kernel/X versions you use?

mod20388, 2023-09-27:
Arch Linux, latest versions (at the moment these are Linux 6.5.4 & X 21.1.8).

Pinglinux, 2023-09-27:
That should work...

I do see hid-generic grabs the device first. The root cause could still be in the kernel. To be certain, can you update your kernel wacom driver by following steps at: https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source?

mod20388, 2023-11-10:
Welp, it seems to work with v1.1.0 (installed from [here](https://aur.archlinux.org/packages/input-wacom-dkms)). So I guess this is one of the "newer tablets".

mod20388, 2023-11-13:
Correction: It is now working when _rebooting_, but it still does not work on a cold boot.

Pinglinux, 2024-05-17:
We can't reproduce it. Do you still face the issue?

mod20388, 2024-05-31:
I got back from holidays recently and _apparently_ it did work for some days. Then I uninstalled input-wacom 1.2.0 to make sure that was really the solution, and well it did stop working. But then I wanted to install it again and now I am getting compile errors (complaints about "strlcpy" even though I do have libbsd installed). I will get back to you and close this when I can install it again and can confirm it works consistently.

Pinglinux, 2024-05-31:
You are probably running a kernel 6.6 or later. Please build the kernel driver from the git repo: https://github.com/linuxwacom/input-wacom/wiki/For-Developers%3A-input-wacom-development-process, which has a workaround for the issue.

mod20388, 2024-06-05:
OK, I managed to get it built. However, the problem remains. Out of 7 boots I got 4 where the tablet was not functional on startup. The driver was loaded in both cases.

Pinglinux, 2025-04-07:
@mod20388 Please follow the links at [1] to capture the input events when the tablet stops functional and share the output here. I wonder if we still get events from the kernel.

[1]  https://github.com/linuxwacom/input-wacom/wiki/Capturing-a-HID-Recording

mod20388, 2025-04-08:
I'd very much like to be of help but in the meantime I've switched to opentabletdriver, which works flawlessly.