linuxwacom/xf86-input-wacom #318: CTH-470 Touch works, stylus does not

Quaxo76, 2023-09-30:
I just bought a CTH-470 to replace my old Intuos. The 470 is a Pen + Touch device.
The touch panel works as intended. But the stylus does nothing at all.
The "Wacom tablet finder" does correctly detect it as "Wacom Bamboo Pen & Touch", with all the expected devices - pad, pen stylus, pen eraser, finger touch.
On the graphic tablet's settings page, I can see no obvious way to enable or disable the stylus.
On the "touch" tab there is a button to enable or disable touch, and this works - if I uncheck it, the touch is disabled, but then the tablet does nothing at all.

Output of `xinput`:
```⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ MOSART Semi. 2.4G RF Keyboard & Mouse     id=10   [slave  pointer  (2)]
⎜   ↳ MOSART Semi. 2.4G RF Keyboard & Mouse Consumer Control    id=11   [slave  pointer  (2)]
⎜   ↳ Wacom Bamboo 16FG 4x5 Pad pad             id=17   [slave  pointer  (2)]
⎜   ↳ Wacom Bamboo 16FG 4x5 Pen stylus          id=18   [slave  pointer  (2)]
⎜   ↳ Wacom Bamboo 16FG 4x5 Pen eraser          id=19   [slave  pointer  (2)]
⎜   ↳ Wacom Bamboo 16FG 4x5 Finger touch        id=20   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Power Button                              id=7    [slave  keyboard (3)]
    ↳ Sleep Button                              id=8    [slave  keyboard (3)]
    ↳ MOSART Semi. 2.4G RF Keyboard & Mouse     id=9    [slave  keyboard (3)]
    ↳ MOSART Semi. 2.4G RF Keyboard & Mouse System Control      id=12   [slave  keyboard (3)]
    ↳ NOVATEK USB Keyboard                      id=13   [slave  keyboard (3)]
    ↳ NOVATEK USB Keyboard System Control       id=14   [slave  keyboard (3)]
    ↳ NOVATEK USB Keyboard Consumer Control     id=15   [slave  keyboard (3)]
    ↳ MOSART Semi. 2.4G RF Keyboard & Mouse Consumer Control    id=16   [slave  keyboard (3)]
```

Output of `libwacom-list-local-devices`:
```
devices:
- name: 'Wacom Bamboo Capture'
  bus: 'usb'
  vid: '0x056a'
  pid: '0x00de'
  nodes: 
  - /dev/input/event29
  - /dev/input/event28
  - /dev/input/event27
```

I'm running Kubuntu 22.04 LTS.

jigpu, 2023-10-13:
1. Please run `sudo apt install libinput-tools` to install a debug tool
2. Run `sudo libinput record`, type in the device number for the Pen, and press Enter
3. You should see a bunch of output scroll past and then pause
4. Touch the pen to the tablet. Is even more output printed out when you do this?
5. Press CTRL+C to quit the recording

If there was additional output at step 4, please copy/paste all of the output into a reply for review.

If there was **no** additional output at step 4, please try the following:

1. Run `ls /sys/bus/hid/drivers/wacom/0003\:056A\:00DE.*/hidraw`
2. A device name like `hidraw3` (or some other number) should be printed out
3. Run `sudo hexdump -C /dev/<name>` (replacing <name> with the device name found previously)
4. Touch the pen to the tablet. Is any output printed out?
5. Press CTRL+C to stop the test.

If there was output at step 4, please let us know.

If there was no output on step 4, it may be a hardware problem or hardware compatibility issue. The CTH-470 is quite an old device, so I assume you bought it used? If so, did it come with a stylus and does the stylus look like the one in the image below? Bamboo tablets generally aren't compatible with pens from earlier or later generations, so it is important to make sure the one you are using is compatible.

[image]

Quaxo76, 2023-10-14:
Thank you for your reply.
I tried method 1, and at first it looked like nothing was happening. Then I "fiddled" a bit with the pen, dragging it around a bit, pushing harder then softer and so on... and occasionally I would get a reading. I'm pasting the contents here, but now I'm suspecting a failing hardware or something like a broken/cold solder point or something. I'll try to find a Windows computer to test the tablet there.

```
Recording to 'stdout'.
# libinput record
version: 1
ndevices: 1
libinput:
  version: "1.20.0"
  git: "unknown"
system:
os: "ubuntu:22.04"
kernel: "5.15.0-86-generic"
dmi: "dmi:bvnAmericanMegatrendsInc.:bvrF14:bd11/13/2021:br5.12:svnGigabyteTechnologyCo.,Ltd.:pnZ370HD3P:pvrDefaultstring:rvnGigabyteTechnologyCo.,Ltd.:rnZ370HD3P-CF:rvrDefaultstring:cvnDefaultstring:ct3:cvrDefaultstring:skuDefaultstring:"
devices:
- node: /dev/input/event26
  evdev:
    # Name: Wacom Bamboo 16FG 4x5 Pen
    # ID: bus 0x3 vendor 0x56a product 0xde version 0x110
    # Size in mm: 147x92
    # Supported Events:
    # Event type 0 (EV_SYN)
    # Event type 1 (EV_KEY)
    #   Event code 320 (BTN_TOOL_PEN)
    #   Event code 321 (BTN_TOOL_RUBBER)
    #   Event code 330 (BTN_TOUCH)
    #   Event code 331 (BTN_STYLUS)
    #   Event code 332 (BTN_STYLUS2)
    # Event type 3 (EV_ABS)
    #   Event code 0 (ABS_X)
    #       Value           0
    #       Min             0
    #       Max         14720
    #       Fuzz            4
    #       Flat            0
    #       Resolution    100
    #   Event code 1 (ABS_Y)
    #       Value           0
    #       Min             0
    #       Max          9200
    #       Fuzz            4
    #       Flat            0
    #       Resolution    100
    #   Event code 24 (ABS_PRESSURE)
    #       Value           0
    #       Min             0
    #       Max          1023
    #       Fuzz            0
    #       Flat            0
    #       Resolution      0
    #   Event code 25 (ABS_DISTANCE)
    #       Value           0
    #       Min             0
    #       Max            31
    #       Fuzz            1
    #       Flat            0
    #       Resolution      0
    # Properties:
    #    Property 0 (INPUT_PROP_POINTER)
    name: "Wacom Bamboo 16FG 4x5 Pen"
    id: [3, 1386, 222, 272]
    codes:
      0: [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15] # EV_SYN
      1: [320, 321, 330, 331, 332] # EV_KEY
      3: [0, 1, 24, 25] # EV_ABS
    absinfo:
      0: [0, 14720, 4, 0, 100]
      1: [0, 9200, 4, 0, 100]
      24: [0, 1023, 0, 0, 0]
[... 101 lines cut ...]
    - [  0, 120022,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +8ms
  - evdev:
    - [  0, 128018,   1, 330,       0] # EV_KEY / BTN_TOUCH                 0
    - [  0, 128018,   3,  24,       0] # EV_ABS / ABS_PRESSURE              0 (-530)
    - [  0, 128018,   3,  25,      31] # EV_ABS / ABS_DISTANCE             31 (+5)
    - [  0, 128018,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +8ms
  - evdev:
    - [  0, 166006,   3,  25,       0] # EV_ABS / ABS_DISTANCE              0 (-31)
    - [  0, 166006,   1, 320,       0] # EV_KEY / BTN_TOOL_PEN              0
    - [  0, 166006,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +38ms
  - evdev:
    - [  1, 103978,   3,   0,   14720] # EV_ABS / ABS_X                 14720 (+14720)
    - [  1, 103978,   3,   1,    9200] # EV_ABS / ABS_Y                  9200 (+9200)
    - [  1, 103978,   3,  25,      29] # EV_ABS / ABS_DISTANCE             29 (+29)
    - [  1, 103978,   1, 320,       1] # EV_KEY / BTN_TOOL_PEN              1
    - [  1, 103978,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +937ms
  - evdev:
    - [  1, 111988,   1, 330,       1] # EV_KEY / BTN_TOUCH                 1
    - [  1, 111988,   3,  24,     571] # EV_ABS / ABS_PRESSURE            571 (+571)
    - [  1, 111988,   3,  25,      28] # EV_ABS / ABS_DISTANCE             28 (-1)
    - [  1, 111988,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +8ms
  - evdev:
    - [  1, 119972,   3,  24,     570] # EV_ABS / ABS_PRESSURE            570 (-1)
    - [  1, 119972,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +8ms
```

whot, 2024-03-03:
Apologies, this slipped through. So based on the kernel output above the pen does send events. But given that you said 

> Then I "fiddled" a bit with the pen, dragging it around a bit, pushing harder then softer and so on... and occasionally I would get a reading

This means the HW is broken. The events recorded here are the ones that come out of the kernel and the kernel does little but convert and forward them on (for extra piece of mind you can use [hid-recorder](https://gitlab.freedesktop.org/libevdev/hid-tools/) which prints the events as they come in from the device with (virtually) no processing whatsoever. But that won't change the outcome much - if you don't get events here as soon as you touch the tablet you have a broken stylus or tablet device.

closed 2024-03-03 by whot