linuxwacom/xf86-input-wacom #319: Wacom Intuos: Straight lines at the ends of strokes

lily-00, 2023-10-15:
Any brush stroke I draw with my Wacom Intuos CTH-690 has a straight line at the end.

[image]

I use Arch Plasma but it happens everywhere I tested, specifically Plasma and GNOME on Arch, Debian 12.2.0 and Ubuntu 22.04.3, in all drawing applications I tested, specifically Krita, Aseprite, Blender and the GNU Image Manipulation Program.

The screenshot above was taken on Plasma with the “raw sample rate” slider in the Plasma tablet settings set to the default 4. This is how it looks everywhere by default, including on GNOME. But changing this value changes the behaviour of the straight end. Setting it to 1 turns the end into a bigger straight line going backwards to where the pen used to be a moment ago, so parallel to the stroke if I draw a straight line and at bigger angles if I draw a curved line. Increasing the value makes the straight end shorter and more aligned with the direction of the brush stroke, probably because of increased smoothing and ending the stroke earlier.

[image]

The longer straight end going backwards when the slider is set to 1 (which I believe means no smoothing) also happens when using OpenTabletDriver on X11 or Wayland, so both xf86-input-wacom and OpenTabletDriver may have the same issue. The ends of strokes are perfectly fine using Wayland with libinput, however.

See also my thread at https://krita-artists.org/t/wacom-intuos-analysis-of-straight-lines-at-the-beginnings-or-ends-of-strokes/75702

lily-00, 2023-10-15:
According to the linked OpenTabletDriver issue, this is a firmware bug in xxx-x90 series Wacom tablets and it can be worked around in OpenTabletDriver by using a plugin that removes the bugged input values from the end of strokes (“Remove Tail Pressure Reports”). That solves the issue for me but it might still be useful for others to know if it’s possible to do this with xf86-input-wacom as well.

Pinglinux, 2025-01-03:
If libinput fixed the issue, it should be used.

closed 2025-01-03 by Pinglinux