linuxwacom/xf86-input-wacom #322: pointer area is not a rectangle

petaflot, 2023-11-03:
in absolute mode, it is clear that the corners of the pointer Area are not those of a rectangle : there is a small but noticeable offset that makes to tool useless for production purposes, and therefore just a fancy toy.

I could not find an option to calibrate the "corners" of the tablet/sensor, does it even exist?

petaflot, 2023-12-02:
tablet has no display, just a black surface with a number of white dots arranged in an oddly-sized non-square grid
```
I: Bus=0003 Vendor=056a Product=037a Version=0110
N: Name="Wacom One by Wacom S Pen"
P: Phys=usb-0000:00:1a.0-1.6.4/input0
S: Sysfs=/devices/pci0000:00/0000:00:1a.0/usb3/3-1/3-1.6/3-1.6.4/3-1.6.4:1.0/0003:056A:037A.000B/input/input30
U: Uniq=1GA00L1140859
H: Handlers=mouse0 event18
B: PROP=1
B: EV=b
B: KEY=1c03 0 0 0 0 0
B: ABS=3000003
```

jigpu, 2023-12-05:
In that case, can you share more about your test setup? I don't understand what you mean by the corners having an offset, unless you mean that the sensor is offset from the white dots --- which is a manufacturing issue, not a driver issue.

petaflot, 2023-12-06:
ok.. si I was planning on going for the more general case (case 4) but even this is not the most general case because some input coordinates have a value of `0`.

does the driver report negative (off-screen) values? this would really help.

Pinglinux, 2024-05-17:
The tablet doesn't report negative values. Do you get the correct maximum (x,y) on the edges of the tablet? If those values are correct, the issue would be in the system's display mapping process.

closed 2025-01-03 by Pinglinux