linuxwacom/xf86-input-wacom #323: pointer position depends on stylus angle

petaflot, 2023-11-03:
I noticed that the pointer's position depends on the angle of the stylus (pretend there is a laser pointer inside the stylus, the pointer will go away from the dot relative to the position of the stylus, normal to the tablet's surface ; ie. the pointer is drawn below the stylus the more acute the angle is which would make sense considering the sensed item (is it a coil? magnet?) is not located exactly at the tip). this also happens when the stylus is barely touching the tablet (I added some double-sided tape to make sure the stylus is not moving relative to the tablet)

can this be compensated?

jigpu, 2023-11-30:
I would suggest using the normal calibration procedure to correct this kind of error. It should reduce the amount of offset seen, as long as the stylus is held at a relatively constant angle (since the calibration is static and does not adjust based on e.g. tilt angle).

For my own curiosity, can you share what tablet you are using? The sensor firmware contains calibration data that normally eliminates the vast majority of this particular type error. In my experience this was much more of a problem with older devices that either were incapable of, missing, or had worse-quality calibration tables. I can't say I've really noticed a significant tilt-correlated position change with recent devices, however.

petaflot, 2023-12-02:
@jigpu it is "One by WACOM" model n° CTL-472

can you point me to the "normal calibration procedure" ? I was not able to find such a thing

jigpu, 2023-12-05:
I was expecting this to be a display tablet; we don't have a calibration procedure documented for non-display tablets. There's not even really any good way I can think to calibrate a non-display tablet since there is no way to know what X/Y values "should" be expected from the tablet for some arbitrary pen position. Even with some absolute reference (e.g. the dots on the tablet surface) it isn't guaranteed that the top-left dot "should" report (0,0) for instance.

You can try tweaking the 'Area' property to improve the calibration. Here's the best guidance I can give for what to do:
1. Run `xsetwacom set <id> area -1 -1 -1` to reset the tablet to the full active area
2. Run `xsetwacom get <id> area` to see the bounds of the full active area. It will be four numbers `<x1> <y1> <x2> <y2>`
    * The four numbers provide the hardware coordinates that will be treated as the logical top-left and bottom-right corners
    * For example `0 0 16000 9000` means (0,0) and (16000,9000) are the logical corners
3. Run `xsetwacom set <id> area <x1> <y1> <x2> <y2>` with a slight modification to one or more of the four values
    * If you increase the value of `x1` (or `y1`) (e.g. to `200`) then at any given physical pen position, the cursor should be moved slightly further left (or further up).
    * If you decrease the value of `x1` (or `y1`) (e.g. to `-200`) then at any given physical pen position, the cursor should be slightly further right (or slightly further down).
    * If you increase the value of `x2` (or `y2`) (e.g. to `16200`) then at any given physical pen position, the cursor should be moved slightly further left (or further up).
    * If you decrease the value of `x2` (or `y2`) (e.g. to `15800`) then at any given physical pen position, the cursor should be moved slightly further right (or further down).
5. Repeat step 3 until you get a calibration that works for your needs. If you need to reset, go back to step 1.

If you adjust `x1` and `x2` (or both `y1` and `y2`) in the same direction by the same amount, it will shift the location of the active area by that amount. If you change the value of `x2 - x1` (or the value of `y2 - y1`) from the default then it will affect the scale factor of the X (or Y) axis.

For reference, I believe the CTL-472 has a hardware resolution of 100 points/mm.

petaflot, 2023-12-06:
unfortunately this doesn't apply here, neither does it to #322 because all 4 corners would need to be defined.

while playing with Gimp, I noticed that some devices apparently have parameters such as orientation and rotation, AFAIK my device does not (but I may be wrong)