linuxwacom/xf86-input-wacom #325: Uneven curly lines

Svampebob1, 2023-11-09:
Hi,
I have a Lenovo Yoga X13 gen. 1.

My problem is uneven curly lines, when I use the stylus to draw diagonal lines. Everything is fine if I draw horizontal or vertical lines.

I've followed the guide on calibration, but this doesn't seem to change anything.
Please see the attached image. The second line is drawed with a ruler.
[image]

I'd be greatful for any suggestion on how to fix this.

Svampebob1, 2023-11-09:
Ps! The more pressure I apply, the more curly the lines become. That's what happened with the second line.

whot, 2023-11-10:
Quick google [suggests this may be a hw issue](https://forums.lenovo.com/t5/ThinkPad-X-Series-Laptops/Extreme-Pen-Jitter-on-X13-Yoga-Gen1/m-p/5039304?page=2)

Svampebob1, 2023-11-10:
> Quick google [suggests this may be a hw issue](https://forums.lenovo.com/t5/ThinkPad-X-Series-Laptops/Extreme-Pen-Jitter-on-X13-Yoga-Gen1/m-p/5039304?page=2)

You are right. I booted Windows and it is the same problem there.
For other people having the same problem, there is a firmware update that should fix the problem.

closed 2023-11-10 by whot