linuxwacom/xf86-input-wacom #327: Wacom touch locks whole screen on Ubuntu 23.10

psycotica0, 2023-11-24:
I'm having a problem that can best be described by watching [this screen recording](https://www.youtube.com/watch?v=hcOapETnvNI) I made to show the issue. But the gist is that switching between windows when using the tablet, but not the mouse, seems to lock up the cursor system-wide in some way. But then in some apps, but not all, every click locks things up in the same way even if they already have focus. Alt+Tabbing away from an app appears to unlock it. Oh, it also may be worth noting that when the cursor locks up I can still see the cursor moving on my screen when I continue to use the tablet, it's just that nothing responds to the cursor. But interestingly, the screen recording (OBS Studio) did not seem to see the cursor moving. Interesting...

I'm not sure if this is an input-wacom issue, or an X.Org issue, or a gnome-shell issue, or what, but you folks probably have a better sense of the stack than I do, so if this isn't an input-wacom issue, I'd appreciate some help in knowing who to send this issue to next.

My system:
Distro: Ubuntu 23.10
Kernel: Linux 6.5.0-13
Gnome: 45
xf86-input-wacom version: 1.2.0
X11: X.Org 7.7

whot, 2023-11-24:
Most likely [this mutter bug](https://gitlab.gnome.org/GNOME/mutter/-/merge_requests/3393)

psycotica0, 2023-11-24:
Ooh! I'm _really_ glad I asked you folks, because I've never even _heard_ of mutter 😛

Ok, I guess I'll watch for that fix to hit my distro and see if that fixes it! Thanks!

patriciopenamunoz, 2023-11-26:
I confirm that I have been able to replicate the same error.

This is my system:
Distro: Fedora Linux 39 (Workstation Edition)
Kernel: Linux 6.5.12-300
Gnome: 45.1
xf86-input-wacom version: 1.2.0
X.Org X Server: 1.20.14
X Protocol Version: 11

I have a "Wacom Intuos Pro M"

psycotica0, 2023-12-09:
I've tested mutter 45.2, and it appears to have solved my issue. Thanks for pointing me in that direction!

closed 2023-12-09 by psycotica0

jbrepogmailcom, 2024-01-13:
Hello. I have installed Ubuntu 23.10 and wacom  stylus did not work correctly on ThinkPad Yoga. I have seen that mutter 45.2 only worked temporarily and there may be fix recently: https://bugs.launchpad.net/ubuntu/+source/mutter/+bug/2043027

However, there are no clear instructions how to apply the fix in normal Ubuntu 23.10 installation

psycotica0, 2024-01-13:
Hello! I'm not the author of the patch, but there are a few answers to "how do I get the fix":

1) Just wait. This is the easiest one. Assuming you have `mantic-updates` in your apt sources (which I believe is the default), then you will get the package in your normal system update once it's been in `mantic-proposed` for long enough to be considered stable. [This update](https://bugs.launchpad.net/ubuntu/+source/mutter/+bug/2043027/comments/16) says "The updated package will be released to -updates after the bug(s) fixed by this package have been verified and the package has been in -proposed for a minimum of 7 days", which means that it should be automatically available to you in a week-ish if you do nothing, and aren't in a rush.
2) You can add the `mantic-proposed` source to your apt and test the proposed package that way. Probably not what you want to do, though, because unless you do some configuring it will probably bring in more proposed packages than you like, and may make the system unstable for you. But technically an option!
3) You can use [this blog post](https://jeremy.bicha.net/2022/03/29/how-to-install-a-bunch-of-debs/) to download the packages from this bug only, and replace the ones on your system with them. This is likely the easiest if you want to try sooner than "just waiting".

Also, I'll mention I used option 3 to test the first version of the package, and while there was a regression in some other part of the system, it wasn't actually something wrong with the Wacom part. It was just another change that was rolled into the same release.

Good luck!