linuxwacom/xf86-input-wacom #334: Assign buttons with xsetwacom

mpaglia0, 2024-01-18:
Sorry but I am getting crazy in assign some buttons to my Huion tablet with xsetwacom.
In particular cannot assign `+`, `-`, `[` and `]`
I read many tutorials and FAQs and understood I have to find the button identifier with `xev`
Doing this I can find, for example, following information for button `+`:
`keycode 35 (keysym 0x2b, plus)`
I tested:
`xsetwacom --set <my_tablet_ID> Button 1 "key 0x2b"` or
`xsetwacom --set <my_tablet_ID> Button 1 "key plus"` or
`xsetwacom --set <my_tablet_ID> Button 1 "keycode 35"` or
`xsetwacom --set <my_tablet_ID> Button 1 "button 35"`
None of these works....
And this is a single button!
For `[` I tested:
`xsetwacom --set <my_tablet_ID> Button 1 "key ISO_Level3_Shift 8"`
and does not work...

whot, 2024-01-18:
`xsetwacom list` shows the device? If not, you're using the xf86-input-libinput driver but installing the wacom driver should do the trick.

After running the xsetwacom command, check `xinput list-props <device name>` and see if you have the action properties and they're referenced by the buttons.

`xsetwacom --set <my_tablet_ID> Button 1 "key plus"` should be the right command, assuming that your current keymap has a `plus` (that's the drawback, we can only resolve things that are in your *current* keymap).

mpaglia0, 2024-01-18:
Ciao @whot  and thank you for your prompt reply!
`xsetwacom list` correctly shows pad and stylus info.
`xinput list-props <device name>` output is quite verbose and I cannot understand it :-(
I am enclosing it.
[output.txt](https://github.com/linuxwacom/xf86-input-wacom/files/13975738/output.txt)

mpaglia0, 2024-01-18:
`xev -event keyboard` output pressing the `+` button
[plus.txt](https://github.com/linuxwacom/xf86-input-wacom/files/13975820/plus.txt)

mpaglia0, 2024-01-18:
Output for `AltGr` button
[altgr.txt](https://github.com/linuxwacom/xf86-input-wacom/files/13975836/altgr.txt)
I need `AltGr` to obtain `[`

whot, 2024-01-22:
> xinput list-props <device name> output is quite verbose

how to read this:
`Wacom Button Actions (326):	"Wacom button action 0" (327), "Wacom button action 1" (328), "Wacom button action 2" (329), "None" (0), "None" (0), "None" (0), "None" (0), "Wacom button action 3" (330), "Wacom button action 4" (331), "Wacom button action 5" (332), "Wacom button action 6" (333), "Wacom button action 7" (334), "Wacom button action 8" (335), "Wacom button action 9" (336), "Wacom button action 10" (337), "Wacom button action 11" (338), "Wacom button action 12" (339)`

This means the property named "Wacom Button Action" (unique id 326) is a list of other properties, in the order of the buttons. When you press the button, the driver goes to the action of the respective button and performs that macro sequence. So to look at what e.g. button 4 does, we need to look at the "Wacom button action 4" (331) property. And that shows:

`Wacom button action 4 (331):	1572873` - that value in hex is `0x180009` which ends up being the [`Xwacom.h`](https://github.com/linuxwacom/xf86-input-wacom/blob/master/include/Xwacom.h) defines:
`AC_KEYBTNPRESS | AC_BUTTON | 9` or, in other words: physical button 4 is mapped to logical button press 9. The driver automatically generates a release when a button is still down at the end of the action. This action is the default (logical buttons 4-7 are reserved and buttons are 1-indexed in X), so our zero-indexed physical button 4 is physical button 5 in X and mapped to logical button 9.

Your button 1 is mapped:
```
 Wacom button action 1 (328):    1114162, 1114138, 65562
```
which is hex `0x110032`, `0x11001a`, `0x01001a`. `0x10000` is the `AC_KEY` action, so we have "key press of code 0x32", "key press of code 0x1a" and "key release of code code 0x1a", with the driver auto-releasing anything still down at the end of the action.

Key codes these days are `(evdev code + 8)` so you can find those in `linux/input-event-codes.h. So 0x32 (decimal 50) is actually kernel event code decimal 42 and that's `KEY_LEFTSHIFT`. 0x1a is decimal 26 so kernel code 18 and `KEY_E`. IOW, that button currently sends shift+e. Note that the various `KEY_FOO` are as the US layout would have them and describe physical keys, so `KEY_Q` is the top-left key even in a french azerty layout.

By the same calculation, button 0 seems to do ctrl+z.

If the xsetwacom command doesn't resolve those correct, could be that code is buggy but at least now you should be able to debug what it's actually setting :)

And as a workaround, you can set those numbers directly:
```
xinput set-prop "HUION PenTablet Pad pad" "Wacom button action 1" 12345 56789 ....
```
(unfortunately xinput does not currently accept hex values)

mpaglia0, 2024-01-23:
Thank you indeed for your kind help and patience .
Problem was caused not by the driver but by a wrong match with tablet buttons...
Now all works fine.
Thank you again!
Maurizio

closed 2024-01-23 by mpaglia0