linuxwacom/xf86-input-wacom #336: Art Pen (KP701E2) pressure interpolation

saldejums, 2024-02-15:
As exactly stated in #187 the Wacom Art Pen (KP701E2) only reports the pressure data every other response.
This is not an isolated issue to a single pen. Have tested this with multiple art pens. I've also talked to others with Art Pens and when I get them to disable the high sensor stabilizers they have set in (painting) software, the pen pressure stepping also appears instantly.

Here is the readout for `libinput record`
[artpen.txt](https://github.com/linuxwacom/xf86-input-wacom/files/14296474/artpen.txt)

Would be great to tie the pressure smoothing to RawSample somehow for ArtPens.
I'd be happy to help out in any way I can.

whot, 2024-02-15:
Can you run `hid-recorder` from [hid-tools](https://gitlab.freedesktop.org/libevdev/hid-tools/) too please? Just to verify this is really in the HID reports and not the kernel somehow dropping reports. Thanks

saldejums, 2024-02-17:
Here you go
[hid-record.txt](https://github.com/linuxwacom/xf86-input-wacom/files/14318753/hid-record.txt)

whot, 2024-02-19:
Thanks, confirming, the pressure only changes on every second event so this is definitely a hardware limitation.

whot, 2024-02-20:
@jigpu, @pinglinx, @skomra, @Joshua-Dickens - do we know if this is a *pen* issue or a tablet issue? i.e. can we expect this behaviour on all tablets?

whot, 2024-02-28:
FTR, it looks like this is a pen issue rather than the tablet, with [this udev-hid-bpf MR](https://gitlab.freedesktop.org/libevdev/udev-hid-bpf/-/merge_requests/34) looking like it's the best solution.

ChengduLittleA, 2024-05-17:
This is `libinput record` for my art pen :thinking: 

[artpen_cintiq_pro24.txt](https://github.com/linuxwacom/xf86-input-wacom/files/15342458/artpen_cintiq_pro24.txt)

Pinglinux, 2024-05-17:
The solution has been merged.

closed 2024-05-17 by Pinglinux