linuxwacom/xf86-input-wacom #339: Remove old detritus from the man page

whot, 2024-03-03:
Low-hanging fruit if someone finds the time: the initial parts of the wacom man page are effectively a decade out of date. We don't recommed using `Section InputDevice` anymore, disabling hotplugging is so niche it's not worth mentioning, etc.

flying-elephant, 2024-04-02:
@whot 
Hi,
I'm working on your request currently and I'd like to make sure before proceeding it.
Do you suppose all of the descriptions related to InputDevice section deleted such as all of the options?
I'd like to make sure of it.

whot, 2024-04-02:
basically: `Section InputDevice` should be `Section InputClass` with the corresponding examples. The whole driver-internal device hotplugging section needs a rework: `inputclass` and input device hotplugging have been the default for 15 or so years now, so we should default to explaining this.

The actual options should be left alone, unless they don't really matter anymore.

PR #357 (2024-10-21): man: discourage use of Section InputDevice
Closes #339
commit 948d0ba8 (2024-10-21): man: discourage use of Section InputDevice

Closes #339
closed 2024-10-21 by whot