linuxwacom/xf86-input-wacom #356: Tilt not working on wired Intuos CTL 4100

pablogila, 2024-10-20:
The tilt does not work at all on Intuos CTL 4100. Attached is the log file.

[sysinfo.rhITjslOoh.tar.gz](https://github.com/user-attachments/files/17451688/sysinfo.rhITjslOoh.tar.gz)

whot, 2024-10-21:
can you run `libinput record` please (might be in `libinput-utils.rpm` or corresponding deb file and *attach* the output here, thanks.

pablogila, 2024-10-21:
Here it is. I got this log file with `sudo apt install libinput-tools` followed by `sudo libinput record > libinput_pen.txt`. I approached the pen vertically and then tilted it to draw a circle.

[libinput_pen.txt](https://github.com/user-attachments/files/17457431/libinput_pen.txt)

whot, 2024-10-21:
Looks like you have the tool 0x862 (ABS_MISC of decimal 2146) and libwacom says that tool doesn't support tilt. Mind you, the tablet doesn't appear to support any other tools anyway, so the driver doesn't even enable tilt for it.

Which means - everything working as intended albeit apparently not as expected :)

closed 2024-10-21 by whot