linuxwacom/xf86-input-wacom #359: unable to configurate buttons for PTH-651

fugata, 2024-11-19:
I tried to use xsetwacom to configurate the buttons for my PTH-651.  Here are the stesp I took:

- use  xsetwacom --list devices  to find the name/ID of my device 
- use  xsetwacom --list modifiers to confirm that I can use "ctrl" and "shift"
- use  xev | grep button  to find the name of the buttons 
- use  xsetwacom --set "Wacom Intuos Pro M Pad pad" button 3 "key ctrl d" to set button 3 to control-D

After I did the above, I opened up an xterm, moved the cursor to the xterm, and then click button 3 --- the xterm then closed, a confirmation that this worked.  But I when moved the cursor to the graphics program (xournalpp, in case that matters) and then clicked button 3, nothing happened -- it's supposed to create a new page, and it worked when I issued "ctrl d" from the keyboard instead of the wacom.

I used   xsetwacom --get "Wacom Intuos Pro M Pad pad" button 3     to find what button 3 actually did, and I got
       key +Control_L +d -d

Is that correct?  I also tried 
     xsetwacom --set "Wacom Intuos Pro M Pad pad" button 8 "key shift ctrl p"
Again button 8 did nothing within the graphics program, and 
     xsetwacom --get "Wacom Intuos Pro M Pad pad" button 8
returned
     key +Shift_L +Control_L +p -p
Is that correct?  Is there a way for wacom to send shift and control instead of shift_L and control_L?  What are all the +/- signs, and why is there an "-p" and "-d" at the end?

Many thanks!

Pinglinux, 2025-01-03:
"+" means a button/key is pressed; while "-" means the button/key is released.

Pinglinux, 2025-06-20:
I assume the issue was resolved by my reply ;).

closed 2025-06-20 by Pinglinux