linuxwacom/xf86-input-wacom #362: tangential pressure stuck on Intuos 3

ec1oud, 2025-01-10:
I have a PTZ-630 with several styli, including an airbrush.  The airbrush  tangential pressure seems to get stuck at one value, maybe the first value that it saw after the tablet has been plugged in and the stylus comes into proximity, or something like that. 

It still works fine on wayland, so the hardware is OK and libinput is ok.

I mostly take care of the wacom support in Qt, and this is the line where it's reading tangential pressure on X11:

https://code.qt.io/cgit/qt/qtbase.git/tree/src/plugins/platforms/xcb/qxcbconnection_xi2.cpp#n1650

It's possible there is some new bug in this Qt plugin, but I doubt it.  I've been testing it often enough over the years.

When running `xinput test` I don't see any numbers changing in response to  the movement of the tangential pressure thumbwheel.

whot, 2025-01-28:
Can you **attach** the output of `libinput record` please so we can have a look at the events, this will make it possible to replay and test locally too. Thanks.

ec1oud, 2025-02-06:
Done.  I tried rolling the wheel both with and without touching the pad.

[recording.gz](https://github.com/user-attachments/files/18695216/recording.gz)

ec1oud, 2025-02-06:
I forgot where I saw version 1.6.4-2 but actually xf86-input-wacom is version 1.2.3-1

whot, 2025-02-07:
> If I switch to libinput in my /etc/X11/xorg.conf.d/99-wacom.conf it doesn't even distinguish the airbrush from a regular stylus. So the wacom driver still needs to be kept working.

ftr please file a bug against [xf86-input-libinput](https://gitlab.freedesktop.org/xorg/driver/xf86-input-libinput/), this should work. Note that the libinput driver isn't fully compatible with the wacom driver so there's a chance that the libinput driver *does* export this information but nothing uses it (e.g. [this mr](https://gitlab.freedesktop.org/xorg/driver/xf86-input-libinput/-/merge_requests/50))

whot, 2025-06-24:
afaict this issue is just for the xf86-wacom driver, the xf86-libinput issue is tangential (just because it was mentioned in the summary0

Pinglinux, 2025-06-24:
Thanks @whot. I see what you mean from the description. I'll test it.