linuxwacom/xf86-input-wacom #363: Buttons only work if stylus isn't near tablet

tristengrant, 2025-03-12:
I'm able to remap my Wacom Intuos Pro M's tablet buttons and they work if my style isn't near the tablet's surface. However, when drawing I'd often need to press a button while using the stylus. 

Is there a setting to fix this?

whot, 2025-03-12:
The tablet "pad" buttons (on the physical tablet) are always connected so they always work.

The buttons on the stylus can only work when the stylus is in proximity of the receiver and those buttons cannot be detected otherwise - this is a HW limitation. They are not BLE buttons like the tail end on some pens.

Assuming I understood your question correctly, this isn't something we can fix.

closed 2025-03-12 by whot

tristengrant, 2025-03-12:
My issue is with the buttons on the tablet. If I put my stylus near the tablet's surface (so I can use the stylus) the buttons don't register anything, however if I move the stylus away the buttons on the tablet do work.

whot, 2025-03-12:
If you run `sudo libinput record` (probably in a `libinput-utils` package or similarly named) against the tablet pad device, do you see events when you press the buttons while the stylus is in proximity? If not then there's a bug in the kernel or it's a hardware issue. If you do, then it's in userspace somewhere.

tristengrant, 2025-03-12:
Thanks, this showed me that it was indeed registering the buttons but not working correctly. However, reinstalling libwacom and xf86-input-wacom seemed to fix it. Probably some issue with the original install.