linuxwacom/xf86-input-wacom #364: You can set the PanScrollThreshold in the .conf but I see no option in the Man page to turn panning on in the first place.

NathanaelGrix, 2025-03-18:
I can't seem to find a way to permanently set pan on. I would just use the xsetwacom command to turn panning on but every time I unplug my drawing tablet from my laptop I have to set it with the command again because it will always just default to pan being off when I plug it back in. It gets pretty annoying because I am moving my laptop a lot so I need to keep unplugging and re-plugging in my drawing tablet. Is there any way around this or is that feature just not currently available?

Pinglinux, 2025-04-07:
@NathanaelGrix you can create a xsetwacom script to start the feature for you when the system starts or when you login [1]. The static .conf file is not straight forward with newer X server on newer Linux distributions.

[1] https://github.com/linuxwacom/xf86-input-wacom/wiki/xsetwacom-example-scripts

NathanaelGrix, 2025-05-15:
I thought it was failing to keep it's settings after hot swapping but after looking at  how those example scripts mention hot swapping support I am starting to think my KDE Tablet daemon might be overwriting my settings every time I re-plug my tablet into my laptop -_-... Not sure how I can shut that off, but whatever I guess that is more of a me issue. Anyway thanks!

closed 2025-05-15 by NathanaelGrix